corecoding/Vitals #5: Three dots icon

medicamin, 2018-08-09:
Hi there,

Could you please change the default three dots icon (...) to something relevant to sensors or monitoring?
i think a tiny ECG of a heart beat is nice

corecoding, 2018-08-09:
This extension is designed so that the menubar shows your crital sensors. When clicking the 3 dots, then a category such as System, then click on Load 1m. This should change the 3 dots to your machine's 1 minute load average. Can you try that and let me know if the dots go away?

medicamin, 2018-08-09:
I didn't know that. Yes it changes to the selected item. But I cannot 
find any way to unselect that item.
beside that, maybe someone (such as me) wants the panel be clean, so he 
or she only needs the menu, not the items being on the panel.

corecoding, 2018-08-11:
I will implement this feature. Stay tuned.

corecoding, 2018-09-02:
Can you guys check out master and help me test single icon mode? The setting is under Preferences

davidrhoderick, 2018-09-03:
I get the system and memory icons plus the numbers, great job!

corecoding, 2018-09-03:
Awesome! Did you test the single icon mode? There is a setting under the cog (preferences) screen.

corecoding, 2018-09-03:
Made some changes. If you have a moment, please do a git pull again and (possibly) reboot to see if fixes it.

corecoding, 2018-09-04:
In the next release (which should be within 15 days) you will be able to deselect all sensors in the menu and will be left with a single icon (with no sensor data). The icon design itself is taken from your gtk theme.

closed 2018-09-04 by corecoding