corecoding/Vitals #7: Hide metric

psukys, 2018-08-10:
Selecting a metric that has a tick doesn't hide it from the panel. Is there a different workflow for that?
Workaround for now: disable -> enable extension and enable wanted metrics to be shown on panel

corecoding, 2018-08-11:
This is definitely not expected! What distro/version are you using?

psukys, 2018-08-11:
I run updated Fedora 28

corecoding, 2018-08-11:
I installed Fedora 28 and Vitals appears to be working properly. If you
have two sensors selected, and deselect one, does it remove from the
menubar? Currently, Vitals requires at least one sensor to be selected, so
if you are trying to deselect the last sensor, it won't deselect. You have
to select another sensor, then select the undesired sensor.

If you have more than 1 sensor, and they still don't deselect, then
something weird is going on. In which case can you look in /var/log/syslog
and search for Vitals to see if there are any errors when you
select/deselect? You can even tail -f /var/log/syslog while you
select/deselect and see if it outputs anything.

psukys, 2018-08-14:
@corecoding you're right. It works as you described, thanks!

closed 2018-08-14 by psukys