corecoding/Vitals #13: How to localize this extension?

amivaleo, 2018-08-19:
Hi!
I saw that you have a "po" folder with two different pot files.
I've been localizing other extensions and I'd like to localize this extension too, in my native language (italian). Where to put the files and which names should I use?

Thanks. :)

corecoding, 2018-08-20:
That sounds great. I’m looking into it and will keep you posted. The extensions I looked at seem to use a locale directory with .po and .mo files. Need to figure out how to set them up and compile them.

corecoding, 2018-08-20:
I have created a branch off of master called localization. Do you mind checking it out and looking at locale/en_US as a template? You probably know better than I, but if copy this into a directory called it and change accordingly, you should be good. Once I can confirm I have given you the ability to translate all strings, i'll merge your changes into master. Thanks a lot!