corecoding/Vitals #17: use placeholder for "Core %d" strings

wbolster, 2018-08-24:
that would help translators :)

corecoding, 2018-08-24:
Is this what you meant? I tried it and it didn't work

msgid "Core %d"
msgstr "Something %d"

wbolster, 2018-08-24:
yes, but your code needs to do the formatting. not sure what the standard js way of formatting is

corecoding, 2018-08-29:
For now the translation files have been updated to support just 'Core' so it is a one-liner instead of 24. This will hopefully help for now. :+1:

wbolster, 2018-09-04:
thanks, but this:

```
label = '%s %d'.format(_('Core'), cpu.substr(3));
```

is not the right approach to localizing strings with placeholders. can you change it to

```
label = _('Core %d').format(cpu.substr(3));
```

instead?

corecoding, 2018-09-04:
That’s what I tried first (code wise). The translation didn’t replace the text as expected.

corecoding, 2018-09-05:
Thank you for the help. To be honest, the way I implemented the fix did feel weird. Can you think of any example extensions I can look at that do similar %d matching? It is slightly different than the singular/plural implementation... right?

wbolster, 2018-09-05:
This is unrelated to singular/plural strings. This is just counting:  ‘core 1’, ‘core 2’, and so on.

My example is correct for this use case.

corecoding, 2018-09-06:
Did some research. It is related to singular/plural when it comes to the code. Went through the Git history of a few projects, and they specially changed _() alias to ngettext. By using this:

label = ngettext("Core %d", "Core %d", cpu.substr(3)).format(cpu.substr(3));

It gives you the ability to set msgstr "Core %d". In the event that a language did have the need to set singular/plural text, that is now an option.

wbolster, 2018-09-06:
sorry, but you are wrong. this string is NOT a format string involving a plural, hence ngettext() should not be used here.

my suggestion is still the correct one.

what leads you to think otherwise?

corecoding, 2018-09-06:
I stand corrected. I tried your original proposal once more and it is working. Thank you.

closed 2018-09-07 by corecoding