corecoding/Vitals #20: hide empty groups

wbolster, 2018-08-27:
currently empty groups show up in the menu, with a expander arrow which actually does nothing. see the second and third item in this example:

[image]

it would be nice if those are just completely left out since there is no information.

corecoding, 2018-08-29:
Today, you can go to Preferences and turn them off. At a later stage, I'll make it so they go off when the system doesn't support Voltage or Fan. The problem is that because the polling is asynchronous, I don't know that certain sensors don't exist until after the menu is drawn. I'll leave this issue open for now.

corecoding, 2019-03-19:
Closing, may open again in the future. Thanks :)

closed 2019-03-19 by corecoding