corecoding/Vitals #29: Not getting fan speed

t4nn3d1n, 2018-09-07:
Running Ubuntu 18.04 w/ Gnome 3.28.2 (ubuntu prepackaged version) on a macbook pro 8,1 (13inch Early 2011)

Vitals is not display fan speed in either drop down and clicking on "fan" does nothing.

the package "psensor" and term-command "sensors" are returning the exhaust fan RPM value.  Other sensors to be working/displaying beautifully.

When I run "PSensor" it gives me a response for the actual sensor being used, which pretty much matches with term-command "sensors":
lmsensor applesmc-isa-0300 exhaust

Let me know if I am posting in the wrong spot or of any more information needed
[image]
[image]

corecoding, 2018-09-07:
Vitals reads sensor values (temp, voltage and fan) from /sys/class/hwmon#/*. If the fan section is empty, then it means Vitals can't find the fan speed input files. The interesting part is that your sensors command is showing fan speed. It may be difficult, but if you can browse the files in /sys/class/hwmon#/* and find out what file is reporting the RPM, it would help me figure out and hopefully add fix to Vitals.

Another option to browsing the above directories is to run strace sensors, which will output what files it is reading for the fan speeds.

t4nn3d1n, 2018-09-07:
/sys/class/hwmon/hwmon2/device/ is the only directory containing any fan labeled files and when I ran strace that was the only hwmon#/ I could read.

Would it be helpful is if I posted the strace and the various fan1_* (input, label, manual, max, min, output, safe) to something like pastebin?

I noticed that fan1_input matches whatever Psensor is reporting, by doing "cat fan1_input"

corecoding, 2018-09-07:
So if I understand correctly, you have a /sys/class/hwmon/hwmon2/device/hwmon/hwmon2/fan1_input but no /sys/class/hwmon/hwmon2/fan1_input?

In all of my test systems, the contents of /sys/class/hwmon/hwmon# matches /sys/class/hwmon/hwmon#/device/hwmon/hwmon# (that is, if a device directory exists).

If you can run:

strace sensors &> sensors.txt

and then send me sensors.txt (attaching to this ticket is fine)

I think this would help me see the decisions sensors is making in traversing the directory structure further.

t4nn3d1n, 2018-09-08:
I have both "/sys/class/hwmon/hwmon2/device/hwmon/hwmon2/device/fan1_input" & "/sys/class/hwmon/hwmon2/device/fan1_input"; sorry for the confusion: I am not used to running/reading strace and only noticed the one path.  For some reason "strace sensors | grep fan" just returned the entire strace.

[image]
[image]
[sensors.txt](https://github.com/corecoding/Vitals/files/2363344/sensors.txt)

corecoding, 2018-09-16:
Can you checkout the branch macbook-support and see if you see your fan speed? Unfortunately I am unable to test it myself, but I think it will work. :)

t4nn3d1n, 2018-09-16:
I am still not getting fan speed, however I would totally understand if I'm not getting how to select a specific branch correctly.  

1) `rm -rf .local/share/gnome-shell/extensions/Vitals@CoreCoding`
2) restarted gnome via "alt-f2", "r"
3) `git clone https://github.com/corecoding/Vitals.git -b macbook-support --single-branch ~/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com`
4) restarted gnome again with same method

corecoding, 2018-09-18:
Those deployment steps look correct. Just a heads up, after testing the feature don't forget to install the plugin again from extensions.gnome.org so you get regular updates.

I did make some more changes. Can you try one more time? this time, the steps are easier

# cd .local/share/gnome-shell/extensions/Vitals@CoreCoding
# git pull
# alt-f2 - r

t4nn3d1n, 2018-09-18:
wow, that command set is way easier

and I have fan speed now, woot

thank you very much for your help

one last thing: if I install from extensions.gnome.org won't that "basically" reset me to the master branch?

NelsonVides, 2019-03-02:
Hi there,
I'm having a similar problem, if you don't mind re-checking this old ticket (or should I open a new case?). I'm on a Dell XPS 15 9570, fan's info is showed on sensor's command, and it's available at hwmon1/, but not in Vitals' panel.

[sensors.txt](https://github.com/corecoding/Vitals/files/2922285/sensors.txt)

corecoding, 2019-03-09:
We can re-open this ticket. Do you have a hwmon1/name file? If not, do you know what file is specifying the name of the fan sensor?

kageurufu, 2019-03-21:
I'm having the same issue, 

```
% sensors
...
thinkpad-isa-0000
Adapter: ISA adapter
fan1:           0 RPM

% ls /sys/class/hwmon/hwmon2/ 
device  fan1_input  name  power  pwm1  pwm1_enable  subsystem  uevent
% cat name
thinkpad
% cat fan1_input
0
% cat pwm1
255
% cat pwm1_enable
2
```

lorddex, 2019-04-11:
> I'm having the same issue,
> 
> ```
> % sensors
> ...
> thinkpad-isa-0000
> Adapter: ISA adapter
> fan1:           0 RPM
> 
> % ls /sys/class/hwmon/hwmon2/ 
> device  fan1_input  name  power  pwm1  pwm1_enable  subsystem  uevent
> % cat name
> thinkpad
> % cat fan1_input
> 0
> % cat pwm1
> 255
> % cat pwm1_enable
> 2
> ```

same here

corecoding, 2019-04-11:
For those of you who cat a fanX_input and see 0, unfortunately, it'll
report 0 in Vitals. Vitals does not know about any specific hardware
configuration. It reads sensor data provided by the Linux kernel.
Sometimes, loading a specific sensor module for your hardware can get more
sensors to appear. Try running 'sensors-detect'.

closed 2019-06-15 by corecoding

setanarut, 2026-10-16:
@corecoding The fan isn't working for me either; I think there was a kernel update. Here is my output.

```bash
❯ cat /sys/class/hwmon/hwmon2/name
nvme
❯ cat /sys/class/hwmon/hwmon3/name
applesmc
❯ cat /sys/class/hwmon/hwmon3/fan1_input 
2154
❯ cat /sys/class/hwmon/hwmon3/fan2_input
2001
❯ uname -r                              
7.3.0-9-generic
❯ ls  /sys/class/hwmon/hwmon3/     
device	     fan2_label	  pwm2_enable	temp13_input  temp17_input  temp20_input  temp24_input	temp28_input  temp31_input  temp35_input  temp39_input	temp42_input  temp4_input  temp8_input
fan1_input   fan2_max	  subsystem	temp13_label  temp17_label  temp20_label  temp24_label	temp28_label  temp31_label  temp35_label  temp39_label	temp42_label  temp4_label  temp8_label
fan1_label   fan2_min	  temp10_input	temp14_input  temp18_input  temp21_input  temp25_input	temp29_input  temp32_input  temp36_input  temp3_input	temp43_input  temp5_input  temp9_input
fan1_max     fan2_safe	  temp10_label	temp14_label  temp18_label  temp21_label  temp25_label	temp29_label  temp32_label  temp36_label  temp3_label	temp43_label  temp5_label  temp9_label
fan1_min     fan2_target  temp11_input	temp15_input  temp19_input  temp22_input  temp26_input	temp2_input   temp33_input  temp37_input  temp40_input	temp44_input  temp6_input  uevent
fan1_safe    name	  temp11_label	temp15_label  temp19_label  temp22_label  temp26_label	temp2_label   temp33_label  temp37_label  temp40_label	temp44_label  temp6_label
fan1_target  power	  temp12_input	temp16_input  temp1_input   temp23_input  temp27_input	temp30_input  temp34_input  temp38_input  temp41_input	temp45_input  temp7_input
fan2_input   pwm1_enable  temp12_label	temp16_label  temp1_label   temp23_label  temp27_label	temp30_label  temp34_label  temp38_label  temp41_label	temp45_label  temp7_label
❯ ls  /sys/class/hwmon/hwmon2/
device	name  power  subsystem	temp1_alarm  temp1_input  temp1_label  uevent
```

## Live monitor in terminal

```
❯ while true; do
  echo "$(date +%T) fan1=$(cat /sys/class/hwmon/hwmon3/fan1_input) fan2=$(cat /sys/class/hwmon/hwmon3/fan2_input)"
  sleep 1
done
21:47:33 fan1=2155 fan2=2000
21:47:34 fan1=2171 fan2=2001
21:47:35 fan1=2157 fan2=2000
21:47:36 fan1=2154 fan2=2001
21:47:37 fan1=2155 fan2=2001
21:47:38 fan1=2169 fan2=2000
21:47:39 fan1=2155 fan2=2001
21:47:40 fan1=2154 fan2=1985
```

setanarut, 2026-10-24:
@corecoding Please reopen the issue. The method for reading fan data has changed as of kernel 7.3. https://github.com/linux-on-mac/mbpfan/pull/273

corecoding, 2026-10-25:
This issue is 8 years old, and it is a different issue. Regarding the pull request, do you have a link (PR, discussion) on 7.3 changing the user space? Linus does not allow the kernel to break user space, so it would be crazy to me if they did.

setanarut, 2026-10-25:
It’s not a different issue; I have Vitals installed, and it doesn’t read the fan speed. I didn’t want to open a new issue with the same title for this reason. I have no idea why Vitals fails to detect the fan sensors. With the 7.3 kernel update, the directory for reading fan data has changed—just thought I’d mention that. Similarly, Vitals also fails to detect the AMD and Intel GPUs. (MBP 13.3, 2016)

[image]

In my system, the paths are like this

### CPU Package id 0 temp

```
/sys/class/hwmon/hwmon1/temp1_input
```

### AMD GPU temp

```
/sys/class/hwmon/hwmon3/temp18_input
```

### Fan speed

```
/sys/class/hwmon/hwmon3/fan1_input
/sys/class/hwmon/hwmon3/fan2_input
/sys/class/hwmon/hwmon3/fan1_min
/sys/class/hwmon/hwmon3/fan1_max
/sys/class/hwmon/hwmon3/fan2_min
/sys/class/hwmon/hwmon3/fan2_max
```