corecoding/Vitals #30: Object.value and String.includes is not a function

vjrj, 2018-09-13:
This extension fails in Gnome 3.22.2 because Object.value and String.includes is missing.

If this helps to other, using some [polyfill](https://developer.mozilla.org/es/docs/Web/JavaScript/Referencia/Objetos_globales/String/includes#Polyfill) here and there, like:
```
if (!String.prototype.includes) {
  String.prototype.includes = function(search, start) {
    'use strict';
    if (typeof start !== 'number') {
      start = 0;
    }
    
    if (start + search.length > this.length) {
      return false;
    } else {
      return this.indexOf(search, start) !== -1;
    }
  };
}
```
and 
```
if (!Object.values) Object.values = o=>Object.keys(o).map(k=>o[k]);
```

I can run the extension.

vjrj, 2018-09-13:
Also I used this other [polyfill](https://developer.mozilla.org/es/docs/Web/JavaScript/Referencia/Objetos_globales/Math/max#Example:_Using_Math.max):
```
function getMaxOfArray(numArray) {
  return Math.max.apply(null, numArray);
}
```
instead of `Math.max(...vals)` in `values.js`.

corecoding, 2018-09-14:
Thank you so much for taking the time to fix and supply code. Do you think you can create a pull request by chance? This way the code goes exactly where it should. Assuming that it should go in extension.js above the class but because I don’t have a 3.22 test environment, I can’t confirm it works.

corecoding, 2018-09-15:
Thank you for the PR! I merged your code into master and was able to install Ubuntu 16.04 with Gnome 3.18 for testing. I know you are running 3.22 so you might not be affected by this, but by chance have you tested toggling options in the Preferences dialog? I am curious if they actually save in 3.22.

Also, would you mind pulling master and testing a few tweaks I made? It works for me (except what I outlined above) but I just want to make sure before I commit to Gnome Extensions.

corecoding, 2018-09-15:
Oh, one more question. Was your network section reporting properly?

vjrj, 2018-09-16:
@corecoding you are right:
- preferences don't save
- network gives wrong values

corecoding, 2018-09-21:
Can you checkout the master branch, press alt-f2 then r (enter) and see if network is fixed, and preference saving?

vjrj, 2018-09-21:
Yes, now works as expected. 

Thanks!

corecoding, 2018-10-01:
Thanks for your help.

closed 2018-10-01 by corecoding