corecoding/Vitals #42: Warning at start start on gnome-shell 3.30

terencode, 2018-09-28:
``Some code called array.toString() on a Uint8Array instance. Previously this would have interpreted the bytes of the array as a string, but that is nonstandard. In the future this will return the bytes as comma-separated digits. For the time being, the old behavior has been preserved, but please fix your code anyway to explicitly call ByteArray.toString(array).
                                                  (Note that array.toString() may have been called implicitly.)
                                                  0 File.prototype.read/</<() ["/home/terence/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/helpers/file.js":21]``

corecoding, 2018-09-29:
Thank you. What distro are you using that has 3.30?

terencode, 2018-09-29:
Thank you for the extension ;)
Sorry for not mentioning, I'm using ArchLinux.

tingvarsson, 2018-10-09:
@corecoding here is an example of how to fix similar issues and keep backwards comp: https://github.com/ethanwharris/gnome-nvidia-extension/pull/166

corecoding, 2018-10-10:
Thanks tingvarsson!

Are you or terencode able to test? I just push the change to master. :)

terencode, 2018-10-10:
Tested and works, thanks.

closed 2018-10-10 by terencode