corecoding/Vitals #43: Ignore sensors reporting impossible value

terencode, 2018-09-28:
I have a faulty temperature sensor returning -128°C, this is taken in account and thus mess up the average temperature.

corecoding, 2018-09-30:
I have pushed a fix in #44. It will take a little time before it gets into gnome extension but if you want to check it out you can pull the code before then. :) Closing now, will open if the issue is not fixed

terencode, 2018-09-30:
Thanks, but it's still not fixed. I'm using your extension from master.

corecoding, 2018-09-30:
Do you have "Hide zero values" turned on?

terencode, 2018-09-30:
I had it, but disabling it doesn't change anything for the negative values.

corecoding, 2018-09-30:
By chance do you know the full path and input value of the invalid sensor in the /sys/class/hwmon directory?

terencode, 2018-09-30:
I do: 

> cat /sys/devices/platform/it87.656/hwmon/hwmon3/temp3_input
> -128000

corecoding, 2018-10-01:
Can you try one more time? Do a git pull on master then alt-f2 then r
You may have to toggle the checkbox once

terencode, 2018-10-01:
My bad, I forgot to remove the extension installed from the website which was overwriting the new one installed in /usr/share. It's working correctly now.
Your first fix was enough.

closed 2018-10-01 by terencode

badoughatcher, 2019-08-09:
I wanted to re-raise this because nouveau reports a 511 degree tempature constantly, this seems to be a known issue. Can we adjust for this too?

[image]

corecoding, 2019-08-11:
I am hesitant to change temperature values because if there was a bug, and another sensor was misreported, that would be bad. That being said, have a look at /etc/sensors3.conf - this file allows you to offset temperature values. I have plans to parse this file in Vitals.

catto24, 2025-12-18:
hi! yet another re-raise - i have quite a few sensors reporting extremely incorrect temperatures, but i have no idea how to ignore them. they've already been ignored in /etc/sensors3.conf but i had a quick look in the directory for Vitals and being real i have no idea how to program in JavaScript...

[image]

[image]