corecoding/Vitals #54: Me is null error

sergiozia, 2018-11-21:
Hello,
I'm running under Ubuntu 18.04 and when I install your Gnome extension, it works perfectly.
But when I open a new Gnome shell session, Vitals do not work anymore.
Into the Gnome console (Alt-F2 + 'lg') the following error appears for this extension : Me is null
When I uninstall and re-install the extension it works again until I quit my session and re-login.

corecoding, 2018-11-22:
When you say open a new session, what are you clicking or pressing on the keyboard?

sergiozia, 2018-11-22:
What I mean is that the extension works until I leave my user session. Then, every time I reconnect (for example after a reboot of after turned off my PC), it does not work anymore until I un-install and re-install the extension. When the extension do not works, the only message into the console is : Me is null
I said that for an other user configured on the same machine (my wife), your extension installed in its environment always works perfectly. It's curious.

sergiozia, 2018-11-28:
I do not know how I can help you and how tell you what is the file and line number reporting this error.
I can just see the error in the console.

[image]

Is there a log file that could give more information about the error encountered ?
Or is there a mean to debug this in order to know the file and the line number produces this error ?

Perhaps an object that can not be initialized because the service it needs is not started yet ?

sergiozia, 2018-12-05:
For your information, I tried to completely uninstall the required packages (gir1.2-gtop-2.0 and lm-sensors) and then reinstall them, but without success. This bug is still present

corecoding, 2018-12-07:
Is there anything in /var/log/syslog ? if you open the file and search for Vitals - you should see a similar error but with more detail

sergiozia, 2018-12-17:
I'm coming back to you with a new information
I found what prevents Vitals to start properly on my computer.
If I do not start the gnome shell extension "Workspace Indicator", then Vitals starts always properly !
I also encountered exactly the same problem under the same conditions as well as the same error "Me is null" with a similar extension "System Monitor" instead of Vitals.
So I returned to Vitals and no longer start the extension "Workspace Indicator" which is less useful to me.
There is probably an incompatibility between the Vitals extension (and also System Monitor extension) and the Workspace Indicator extension that is present by default in the Ubuntu 18.04 distribution (?)

corecoding, 2018-12-24:
Thanks for this info! The author of that extension is fmuellner - and they have written many other plugins. Kind of weird that it would be causing an issue like this. But I am glad you found out what it is. I was going to look at any potential reported issues, but they are using gitlab and I don't see an issues area.. Wish I could do more than say don't use Workspace Indicator, but I think it is the only option for now...

closed 2018-12-24 by corecoding