corecoding/Vitals #55: Fixed width

ccrisan, 2018-12-02:
How can I make the widget have a fixed width, either entirely or for each information displayed? When displaying the network upload/download, values change often and move the widgets on the top bar, creating an unwanted visual effect.

wolfen351, 2019-11-01:
A suggestion - Use unicode special space characters to pad the width to a specific size

Tharwat96, 2019-11-30:
fixing this will be great if possible or if there is a way to manually do it.

Obihoernchen, 2020-05-20:
The last feature missing to make this the perfect system monitor tool!

corecoding, 2020-05-22:
I have started looking into this feature. Setting width is easy. Harder part is figuring out the proper max width for each item. Even items in the same section have different widths, like the IP Address one.

Obihoernchen, 2020-05-22:
Maybe adding an user input for each column, so users can set a specific width based on their configuration?

wolfen351, 2020-05-24:
Thanks for looking into this one! :)

coffeemug, 2020-05-26:
> I have started looking into this feature. Setting width is easy. Harder part is figuring out the proper max width for each item. Even items in the same section have different widths, like the IP Address one.

I'd just keep the width from shrinking. This way it might occasionally increase, but that's ok. So long as it doesn't jump around back and forth every few seconds, it isn't really a problem.

corecoding, 2020-05-26:
Yes, that was my idea too. However, I can’t find anything out of the box in the St.Label that does this. This means I have to calculate widths manually and keep track of them. That’s not the end of the world, until you add support for different fonts and font sizes. 

Even if I grab the current width and stick it in an array, and check if the new width is smaller than the previous width, it would still cause jumping effect. 

If anyone has any ideas in how to do this (code wise) it would be appreciated.

wolfen351, 2020-05-28:
Just let me enter a hardcoded number input box for the size, I'll choose something that works for my situation.

wolfen351, 2020-05-28:
Or where you have the calculation you are planning on doing, just do it for a "standard" font, and let me manually enter a "scaling factor" to even out any weirdness caused by fonts etc

coffeemug, 2020-05-29:
Yup, that would work for me too. Strongly agree.

blipk, 2020-05-29:
A possible way to determine the width is to loop through each of the label texts, counting the characters then using set_style to set the width in em/ch based on the amount of characters, would also need to add widths for the amount of icons (possibly probing icon_size for each) + an amount of nominal padding to prevent constant small fluctuations.

mkoreneff, 2020-05-30:
Is it possible to set the width using css in StWidget.st_widget_set_style()?

corecoding, 2020-06-06:
I pushed some changes in, which I think are closer to what you are expecting. Let me know what you think. It is on the develop branch.

wolfen351, 2020-06-07:
I had a quick look at the implementation, looking good! I see you went for the 'don't decrease' approach, which is the best one imo

corecoding, 2020-06-07:
I think so too. Unfortunately, I had to create a private var to keep track of the widths. I tried using the current width of the label, but it wasn’t working correctly.

corecoding, 2020-07-04:
Can anyone else verify that this is working for them?

Obihoernchen, 2020-07-09:
Just tested it and Network is fixed width now, but Processor is not. Width differs between single and double digit and thus moves all other figures.

Obihoernchen, 2020-07-20:
Furthermore, if I get figures like `743 KB/s` network speed the `s` is not showing. Looks like the fixed width is slightly too small

wolfen351, 2020-08-24:
I've been using this for a few weeks now, and it's brilliant, working great! Thanks

glebr, 2021-01-30:
Just let the user set the width manually. This is implemented in resource_monitor gnome-shell extension: https://github.com/0ry0n/Resource_Monitor/

kohane27, 2021-10-15:
I'm reporting that with `use fixed widths (beta)` enabled, for the Processor tab, the digit still moves and changes width.

corecoding, 2022-02-24:
The issue was that when you toggle an option in preferences, or login/logout, fixed widths stopped working. This is fixed in the latest release, which should come out tomorrow. If you experience other issues, please feel free to open a new ticket.

closed 2022-02-24 by corecoding

ilavidas, 2022-03-06:
Fixed width seems to be working, thank you!

HubKing, 2023-09-21:
It seems that fixed width setting is not applied to items that is after turning the setting on. I had to disable and re-enable it to apply fixed width to new items. Please check if this is a bug.