corecoding/Vitals #58: Update POT file

pkomur, 2018-12-28:
Please update POT file. Some strings are missed.

corecoding, 2018-12-28:
Can you pull the master branch? I recently updated the pot file. If I am still missing some, please let me know which ones.

pkomur, 2018-12-28:
Missed:
 Core
 Package id
 Uptime
 Used Cycles
 Format
 Bytes
 Bits

corecoding, 2018-12-29:
Thanks for the feedback.

Core  -  already there
Package id  -  This string comes from the system itself.
Uptime - Added.
Used Cycles - renamed to Process Time. Added. 
Format - added.
Bytes - added.
Bits - added.

corecoding, 2018-12-29:
Are you working on adding a translation?

pkomur, 2018-12-29:
Yes, I am.

pkomur, 2018-12-30:
What is the difference between Uptime and Process Time? It looks like the same.

corecoding, 2018-12-30:
Process time is the time the cpu spends working - that is, not idle. https://en.m.wikipedia.org/wiki/CPU_time

corecoding, 2018-12-30:
Thank you so much for your work here. You did more than just the Polish translation. I appreciate it.

closed 2018-12-30 by corecoding