corecoding/Vitals #66: Centos 7 corrupted after install (help wanted)

gpapaioa, 2019-01-11:
Please I need some help. After trying to Restart GNOME Shell (Alt+F2, r, Enter) my system if corrupted and I get a black screen with cursor after Starting Gnome Display Manager.

gpapaioa, 2019-01-11:
After using 3 on grub I managed to log with no GUI. After running startx on terminal, the GUI starts but I only have a black screen with desktop icons. No bars, no panels... Very strange. Reinstalled the GNOME environment but again no luck.

corecoding, 2019-01-12:
I really don't think this is due to Vitals. Were you modifying the code by chance? The only time (that I know of) that can make Gnome Shell unstable is when you try to reference a setting not yet defined in org.gnome.shell.extensions.vitals.gschema.xml (and compiling after).

Anyhow, if you completely remove .local/share/gnome-shell/extensions/Vitals@CoreCoding.com and reboot, Vitals is gone entirely. If your environment still is crashing, it is not due to Vitals. If your environment is working, then I would be very anxious to see any Vitals related errors in your /var/log/syslog

gpapaioa, 2019-01-12:
Its very strange cause I didn't mess with anything, honestly. Just tried to installed it..

I did remove it as you say. No i am using yum update to fix some stuff. I think installing **libgtop2-devel** messed with something

gpapaioa, 2019-01-12:
Everything seems to be back to normal. Thank you very much for your immediate response.

corecoding, 2019-01-12:
What was the fix? If it was the removal of the Vitals directory that fixed the issue, please send me some info from your /var/log/syslog (I can give a command if you need it)

gpapaioa, 2019-01-12:
No Chris i think something got mixed up really bad with the dependencies. Packages got mixed up and broke the gnome shell. Just updated to the latest kernel too (7.6) and everything is back to normal.

closed 2019-03-03 by corecoding