corecoding/Vitals #68: thresholds

corebots, 2019-01-23:
Could you perhaps add an option to set thresholds for certain values so that they come up in the panel only when they reach a set value first, and perhaps turn red or flash when a set extreme is reached?

EXAMPLE: don't display the CPU temp (or display the little icon only, so we're able to click on it to access the drop down), start to display the temp if it reaches 50degrees, turn red if reaches 75degrees. (perhaps flash if it reaches 90degrees)... Same for CPU, Memory, etc... 

But as the first step, even hiding all below certain values would be nice already. This would keep the panel nice and tidy when all is alright with the system while functionality would be preserved. 

Thank you, great extension btw.!

corecoding, 2019-06-02:
Keeping this open because I am interested in this feature. Don't have a lot of time to work on it, but hopefully one day. Maybe someone will create a PR :)

wolfen351, 2019-11-01:
Tagging this one, because I too would love this feature!

corecoding, 2020-07-15:
I believe the correct way to handle this is to parse /etc/sensors3.conf - this reuses existing alerts (if any) and supports custom naming of sensors as well.

guiAlberto, 2021-10-04:
That feature would be great! I use a fairly old computer to work, and sometimes it freezes when reaches full CPU and RAM usage. I usualy keep an eye on those values, but it is easy to distract and realize those values are high only when it's too late.

kgapos, 2022-11-17:
Same here, came to request an alerting feature. No time to work on this either.

T3sT3ro, 2023-01-10:
Additional thoughts and ideas:

> It would be useful for me to monitor the temperature and show an alert if certain threshold is crossed, as well as setup multiple checkpoints when running out of battery. I haven't found any satisfactory native ways of monitoring those things. UPower, for example, can't configure more than 2 levels for battery power — low and critical. Also I found no nice solution for monitoring temperature and alerting me when something is not OK.
>
> This would be very helpful for me, because my hardware fans fail very rarely, and the PC can overheat and without any warning shutdown immediately. Same goes for the battery, which can discharge randomly in a span of a few minutes, and I sometimes dismiss warnings until the battery is critically low.
>
> For battery it would be nice to have "alert when threshold crossed" and "alert every X units after threshold crossed", for example every 3% if below 15%.

hyuri, 2023-01-16:
Yes, notifications would be great. Came back to suggest this. For temperature, memory, battery. That way we don't even have to keep an eye on the system tray, we get notified on-demand.

T3sT3ro, 2023-01-18:
I'm just gonna mention, that if a temperature tracking notification were to be implemented, then it is also worth thinking about some time-series smoothing function to reduce the spikes. Currently lm-sensors can produce anomalies lasting even several seconds showing temperature ranging even up to 50°. It would be useful to have some controls for adjusting the window size for the smoothing as well.

Sanan4li, 2023-07-11:
Waiting for this feature to be added. I also need a memory limit alert because my system gets stuck after 98% memory use. The linux is not as good as windows in memory management.

rwrnet, 2025-12-23:
I created a small functional prototype, and I like to gather some feedback. See screenshot attached.
[image]
I covered temperature and memory-percentage.
The alert threshold is configured in the sensor preferences.
The style is fix for now:  `background-color: rgba(255, 0, 0, 0.3);`

corecoding, 2026-09-05:
added color support. We can open another ticket for other features if there is demand.

closed 2026-09-05 by corecoding