corecoding/Vitals #71: [Feature Request] Only update when opened/clicked

nealot, 2019-01-25:
Hi, thanks for the great extension! For decreasing power consumption, I would like to suggest a setting where the values are updated only when the user clicks the panel icon. I don't display any of the sensors on the panel, so being able to quickly see the system vitals in the menu without needlessly updating every second would be helpful.

corecoding, 2019-03-09:
I'd like to keep this issue open. I actually started work on this but didn't finish. Honestly I am bummed right now, as Gnome Extensions are generally approved every 15 days. For some reason, Gnome hasn't approved any since December. https://extensions.gnome.org/review/

vinay-gujar, 2019-05-18:
@corecoding could you please make it an optional feature as most of us need the real time vitals on how the system is performing. Thanks!

corecoding, 2022-02-18:
This has been added in https://github.com/corecoding/Vitals/commit/9d09f3be7766c2afe8af683f30d6a20c875b9939
If you have no sensors in the menu bar, it will refresh when the menu is opened.
If you have sensors in the menu bar, it will refresh at the specified time.
Currently in develop branch, will be launched within about a month.

closed 2022-02-18 by corecoding