corecoding/Vitals #74: Doesn't load with GnomeShell 1:3.32.0-1 on Arch

Zappo-II, 2019-03-16:
After update of gnome-shell from 3.30 to 3.32 the plugin won't load anymore (stated by shel-extensions configuration)...and won't show up as it did before...

lordSeaworth, 2019-03-17:
I'm having issues as well with this and i consider vitals as one of my required extensions for gnome. Anyway we can expect a fix soon?

corecoding, 2019-03-17:
Thank you for the kind words. Unfortunately, I’m not sure if an update will be available when 3.32 becomes mainstream. I’m personally waiting for it on Ubuntu. The problem is that gnome hasn’t approved any extensions since mid December. Normally they are approved every 10 to 15 days. I have a pending update that has added some functionality, and it is frustrating. https://extensions.gnome.org/review/ You can see that there are dozens of pending extensions... I have tweeted their fan page and I don’t know who else to reach out to. :(

lordSeaworth, 2019-03-17:
Oh. I c. Yeah their approval/update system is something special.

Guess we will have to wait until they approve it then.

Anyways would like to say thank you for your work on the extension.

corecoding, 2019-03-17:
Is there anything special to installing arch Linux with gnome 3.32? If you can give me a tutorial link I’ll install it and get a jump on Ubuntu’s version

Zappo-II, 2019-03-17:
Most comprehensive HowTo I know...

https://m.youtube.com/watch?v=m28veKzJcQ4

Zappo-II, 2019-03-17:
`
Anyways would like to say thank you for your work on the extension.
`
Jeah, me too...

lordSeaworth, 2019-03-17:
@corecoding 

There isn't anything special about installing archlinux or gnome. Once you have installed archlinux, installing gnome is pretty simple.

Best way to install archlinux if to follow the official wiki and not really follow any youtube videos.

https://wiki.archlinux.org/index.php/Installation_guide

For gnome.

https://wiki.archlinux.org/index.php/GNOME

coxackie, 2019-03-17:
Actually, for me (Arch, GNOME 3.32) it loads, but there are issues with the icons, the submenus do not appears, and weirdly so the processor percentage shown gets crazy sometimes.

corecoding, 2019-03-17:
Thanks for the howto. Arch kind of reminds me of the slackware days :) I'll try my best to get this fixed up today. Worst case is you guys can clone the git repo until it is officially available through gnome.

corecoding, 2019-03-17:
I believe I have it fixed. Seems to work in gnome 3.30 as well. Please do the following:

cd ~/.local/share/gnome-shell/extensions
mv Vitals\@CoreCoding.com Vitals\@CoreCoding.com2
git clone https://github.com/corecoding/Vitals.git Vitals\@CoreCoding.com
cd Vitals\@CoreCoding.com
git checkout gnome-3.32

now press alt-f2 and type r <enter>

If you experience crashes of any type, simply delete this directory.

lordSeaworth, 2019-03-17:
@corecoding 

Nice. Just did your instructions and it seems to work again so far!
If I encounter more issues I'll let you know.

Thanks for fixing this.

corecoding, 2019-03-18:
Good to know. The downside is that with a git clone, you likely won't get future updates when the extension does get updated on Gnome Extensions...

coxackie, 2019-03-18:
Thanks a lot - can also confirm it works now, as far as I have tested. There is only one little glitch, at least in my system: the (middle) icon taking you to system monitor is missing. I mean, the button is there, and if clicked it takes you to system monitor, but there is no icon any more. Maybe cause by the icon redesign in 3.32?

Zappo-II, 2019-03-18:
I can confirm that after...

`git checkout gnome-3.32`

...all my issues are fixed now.

Thank you very much for the very timely support and fix for this highly appreciated extension !

closed 2019-03-18 by Zappo-II

corecoding, 2019-03-20:
Good news everyone. Vitals with Gnome 3.32 support was approved. Feel free to ditch your custom git clone's, so you get future updates.