corecoding/Vitals #77: Missing System Monitor icon in Arch

corecoding, 2019-03-19:
Arch Linux has a missing System Monitor icon in Gnome 3.32 mentioned in https://github.com/corecoding/Vitals/issues/74

Zappo-II, 2019-03-20:
I'm just another user but it works for me on Arch with Gnome 3.32...might be some other individual problem...maybe something icon library related, I'm using adwaita...

corecoding, 2019-03-22:
Do you happen to know the package name for adwaita?

Zappo-II, 2019-03-22:
https://www.archlinux.org/packages/?q=adwaita-icon-theme

WoefulDerelict, 2019-05-15:
The SVG resource used for the System Monitor icon is located in the [gnome-icon-theme-symbolic](https://www.archlinux.org/packages/extra/any/gnome-icon-theme-symbolic/) package. The relevant file is /usr/share/icons/gnome/scalable/apps/utilities-system-monitor-symbolic.svg

corecoding, 2019-05-16:
Thank you. Will update README with package name.

closed 2019-05-16 by corecoding