corecoding/Vitals #80: [Feature Request] hide icon

kapcom01, 2019-03-23:
Hello.
With many extension loaded on the panel it starts to run out of space. It would be helpful if there was an option to hide the icons from the panel. For example, I use this extension to watch my CPU temperature, so a simple "50°C" is enough for me. The thermometer icon does not give me any more information.

Thanks.

durapensa, 2020-10-25:
Vitals is a fantastic little GNOME Shell extension! I came here to post essentially the same feature request, and a more specific suggestion for the Sensors column of the Preferences window: either in the same column (if possible) or a new 'Status Icons' column, provide tick-boxes or "light switch" style toggles to toggle the display of individual sensor icons in the status area, and an ability to re-order said icons. Many thanks for working on this extension and for keeping it up-to-date!

KonstantMan, 2021-05-09:
Yeah that would be really nice option to avoid cluttering all the icons togethere

xcodz-dot, 2021-09-19:
Hey there, Any idea if its being worked on or kept for later?

corecoding, 2021-09-20:
I will add this feature to the next release, seems that enough of you want this feature. I missed the 12 thumbs ups. :)

corecoding, 2021-11-04:
This feature is available for testing in the development branch. Please test it and let me know if it is working.

Instructions for beta testing are listed in the README at https://github.com/corecoding/Vitals under Beta Testing

spxak1, 2021-11-07:
> This feature is available for testing in the development branch. Please test it and let me know if it is working.
> 
> Instructions for beta testing are listed in the README at https://github.com/corecoding/Vitals under Beta Testing

Installed and looks great! Thank you.

corecoding, 2021-11-09:
Vitals 46 released with this feature.

closed 2021-11-09 by corecoding