corecoding/Vitals #82: Missing Mhz Text in Arch Linux

zsantos, 2019-04-23:
Hello, I noticed that on Arch Linux the Mhz text does not appear, is it required to install any particular font ? I have a manjaro system with the same extension and it is working perfectly, just wondering if I am missing anything. 

Thank you for your time.

[image]

corecoding, 2019-05-08:
Thanks. I think I will make this change going forward. I was using Unicode characters to save space in the menu bar. Optionally I can add a checkbox but that seems kind of silly.

zsantos, 2019-05-08:
@christhejoker Thanks mate that did it. Have a good one

closed 2019-05-08 by zsantos