corecoding/Vitals #84: [Feature Request] Second Fan

tye-mustafa, 2019-05-14:
i have a Dell 7559 which have two integrated fans. your extension only show one fan.
[image]
on the other side lm-sensors shows both fans and other extensions on gnome also shows both fans in there extensions.
[image]

please add both fans monitor system.

corecoding, 2019-06-02:
Can you check out the develop branch (instructions on the README) by following step 2a then git checkout develop, then restart gnome? The second fan should show Processor Fan 2

corecoding, 2019-06-15:
Closing. Feel free to reply, and if it doesn’t work I’ll reopen the issue.

closed 2019-06-15 by corecoding