corecoding/Vitals #86: Error opening file /sys/class/power_supply/* No such file or directory

terencode, 2019-05-18:
This probably happens because desktop computers do not have batteries.

corecoding, 2019-05-18:
Yes... You are right. I guess I didn't think many people would be using the master branch as opposed to Gnome Extensions. :) Will clean it up this weekend.

WoefulDerelict, 2019-05-19:
There is a PKGBUILD in the AUR that allows users to easily track changes in master. I don't think a large number of users install it that way or to follow the instructions at 2a in README.md; however, as they are published options and both pull from the master branch users were bound to end up using it.

corecoding, 2019-05-20:
Good to know about PKGBUILD. Thanks for the heads up. Going to close this issue. I removed the file missing warnings. No need to write to the disk every interval anyways.

closed 2019-05-20 by corecoding