corecoding/Vitals #89: Fan and Voltage don't work

ddnexus, 2019-06-01:
Fan and voltage are showing up but don't open the submenu with the values when clicked.

corecoding, 2019-06-01:
Can you share a screenshot?

ddnexus, 2019-06-02:
My description was obviously bad. :)
I mean that you get the usual menu that includes Fan and Voltage items, complete with the drop-down arrow. However if you click on them you don't get any drop-down: it works only for the other menu items, not for Fan nor Voltage.

corecoding, 2019-06-02:
That means that your motherboard isn’t setup to give that info to the kernel. Vitals simply relays what is available in the /says and /proc folders. If you type sensors at the terminal, you will likely see that fan/voltage is not there. If it is, let me know and we can dig deeper.

ddnexus, 2019-06-02:
Oh well...
```
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +54.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +54.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +52.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +51.0°C  (high = +100.0°C, crit = +100.0°C)

dell_smm-virtual-0
Adapter: Virtual device
fan1:        3215 RPM
fan2:        3215 RPM

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +25.0°C  (crit = +107.0°C)

iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +37.0°C  

pch_skylake-virtual-0
Adapter: Virtual device
temp1:        +48.5°C  
```

I am surprised since Vitals was working till recently with Fan and Voltage, then I upgraded Ubuntu 19.10, switched to the NVIDIA driver, and received a firmware update from DELL. I also added `pci-noaer` in the grub configuration (since there was a recurrent pci error in the log).

Which of those change do you think may have caused the problem?

ddnexus, 2019-06-02:
I am actually not 100% sure Voltage was working, but when I first installed Vitals I would have noticed it because I tried it thoroughly. I certainly remember that Fan did work, because I checked it quite a few times.

corecoding, 2019-06-02:
You should have fans at least. If you installed sensor modules without rebooting, Vitals won’t see them automatically, but you can try hitting the refresh button.

ddnexus, 2019-06-02:
Lots of reboot since when I noticed the Fan and Voltage not working. Nothing changes after hitting the refresh.

corecoding, 2019-06-02:
This is weird.. To answer your other question, if I had to guess what caused it, I would guess Ubuntu 19.10. I'm guessing the kernel was updated. Can you run:
strace sensors > sensors.txt
and attach that file here?

ddnexus, 2019-06-02:
You probably intended `strace sensors 2> sensors.txt`, because without the `2` it would dump only the output of `sensors`.

Here it is [sensors.txt](https://github.com/corecoding/Vitals/files/3244487/sensors.txt)

serpro69, 2019-06-13:
I have the same issue on ubuntu 18.04. 
Here's the [sensors.txt](https://github.com/corecoding/Vitals/files/3284954/sensors.txt) file,

corecoding, 2019-06-13:
Can you both try going to settings and turn off Hide zero values? Then press alt-F2, and enter r in the box. This will restart gnome-shell.

I have a new laptop at work that actually turns off its fan at lower temps. With the above checkbox on, it causes Vitals to not show the fan by default.

serpro69, 2019-06-13:
Turned off `Hide zero values` and restarted the gnome shell. Still can't see neither voltage nor fan speed info.

CHIGonzales, 2019-06-15:
Not showing on mine as well

corecoding, 2019-06-15:
The issues reported here are different. The OP was talking about Ubuntu 19.10. It is common to have missing fan/voltage as it isn’t supported on all motherboards. Vitals knows nothing about specific hardware but instead reads from /sys and /proc.

You may want to try the develop build and see if it helps.

It is very difficult to test specific hardware. If you can provide a PR that fixes the issue, I’m open to merging it in the repo.

closed 2019-06-15 by corecoding

corecoding, 2019-08-17:
The next version of Vitals may fix this. Again, if it doesn't I'll need a PR. But, fingers cross a bug I recently fixed will fix this as well.

ddnexus, 2019-09-04:
Confirmed! The Fan section reappeared with data after today's upgrade!
Thank you.

How should I check whether the voltage is available on my system?
Thanks

serpro69, 2019-09-05:
Also have the fan section now. But the readings are all zeroes.

corecoding, 2019-09-05:
Both of my laptops show 0 most of the time. It’s when the CPU works hard that the fans turn on. If it’s a desktop, try turning off the hide zero values option.

You can also try the sensors command in the console. It should also say zero when vitals does.