corecoding/Vitals #90: Request: 1. Max temp 2. Public ipv6 support 3. location for ipv4 (Nat. flag or short USA/GB/CA ..)

jackhistoria, 2019-06-02:
You guys did an awsome job there. Finally an extension that works properly. Thanks for that one!

I would like to pull a request. Or in that case three requests:
1. An option for the Maximal temperature detected by any sensors would be awsome.
The average option is great, but sometimes a blocked fan can go up in temperature without
noticing it and having all sensors listed doesn't fit in my task bar.
So getting the highest temperature would be awsome and all out of one extension. Yours! :-)

2. Since I travel a lot with my IP (vpn/ssh tunnel/proxy) it would be awsome
to be able to select from different servers like in the public ip extension
because some servers reply the ip v4 and some the ipv6 so i have to run 
the IP Finder extension too where the development stopped to see both, the ipv4 and the ipv6
(https://extensions.gnome.org/extension/1677/public-ip/)
[image]

3. In that favor it would also be cool to get a location indicator like a US-flag or the letters USA or CA or a flag or both to select additionally to the ipv4 and the ipv6 addresses just to get proof
the vpn is connected to the chosen country wihout curling the location in an extra step.

4. It would be great to have also a local network ip indicator like 192.168.0.8 shown in the bar
too for administrative reasons.

I am just learning some c++ and am actually not able to include my requests in the extension myself but would do the work myself when I developed the skills to do that.
It would be great just to have one extension to get it all not 4 or 5 that may or may not work properly. I actually don't know how difficult it would bee to add all that points - but even one or two of them would be great.
That being said: Thanks in advance guys - keep up with that good work - i really appreciate it!

oliveconstant, 2021-06-25:
Came here to request maximum temperature feature. I think it's important to be able to monitor the highest recorded temperature overall.

corecoding, 2022-07-10:
We should have these requests split out into separate ticket requests.
1) This seems like a neat requests, perhaps @upward4 or @jackhistoria would be willing to create a stand alone request.
2,3) I would be willing to review a PR for this request, but this request takes vitals out of my personal scope.
4) See https://github.com/corecoding/Vitals/issues/254

closed 2022-07-10 by corecoding