corecoding/Vitals #93: [Feature Request] Reorder indicators

terencode, 2019-07-08:
Be able to reorder in the settings window the different indicators.

humanplayer2, 2021-06-30:
The following is a bit off topic, but I hope may help others that check this feature request:

The order of items shown in the top bar are determined by the order they were added, with things added to the right.

masbaehr, 2021-07-09:
Can't believe this has not been implemented. Until it is i'm trying to manually change the order but without success. Any hint? 

I thought changing the array order would be enough but it does not have any effect :(

[image]

corecoding, 2021-07-11:
As @humanplayer2 said, they are ordered in the way they were added. It used to be the case that when adding / removing sensors the menu would stay on the screen, making re-ordering much easier. Great news is that the next version of Vitals is bringing this functionality back.

If I were to add another way to order, how would you propose the UI would look/work?

humanplayer2, 2021-07-11:
Nice! 

I don't think I would use the UI very many times, so I don't think it would matter greatly to me, personally. You could have us assign psiuton numbers as numerical values and I'd be happy.

But since you ask: You could consider a menu partitioned into Shown and Not Shown sensors, with the Not Shown ordered alphabetically, and the Shown allowing drag and drop reordering, possibly indicated by some small handle (I think I've seen a small version of the menu three-line hamburger used as a handle elsewhere, but maybe I misremember and there is some other standard).

eddie-dunn, 2022-10-14:
> If I were to add another way to order, how would you propose the UI would look/work?

I would say being able to drag and drop items is the most intuitive and common way of solving this. I've never worked on any Gnome extension, so I have no idea how much work that would entail, though.

So let's not perfect be the enemy of good. I think just being able to reorder items by, e.g, manually specifying their index would be enough for most people.

dbash13, 2026-10-19:
Same as #451, still no GUI option to reorder items