corecoding/Vitals #94: Storage not shown

ozls, 2019-07-18:
I just discovered this extension and I love it. But unfortunately, the storage is not showing any kind of drop down menu. I saw issue #45 and it looks like my problem, but restarting the shell did nothing. Is there something that I am missing ? I have Ubuntu 18.04.2 with GNOME 3.28.2.

corecoding, 2019-07-19:
Please install the packages mentioned in the README file.

ozls, 2019-07-19:
Oops. Thank you for the answer, I'm very sorry for being a non-readme-reading asshole :-D Have a nice day !

closed 2019-07-19 by ozls

liver, 2023-08-08:
Storage not showing...
[image]

liver, 2023-08-14:
> Storage not showing... [image]

solution:
sudo apt install gir1.2-gtop-2.0 lm-sensors

than, reboot system

youyoumu, 2025-06-01:
for NixOS users, you should install the extension from [gnomeExtensions.Vitals](https://search.nixos.org/packages?channel=unstable&from=0&size=50&sort=relevance&type=packages&query=gnomeExtensions.Vitals) package. It has the necessary patch to find libgtop