corecoding/Vitals #96: New version lacks sensors

mechano, 2019-08-01:
Downloaded new version 24 jul 19, and now no CPU and chassis fan, only GPU fan.
Also lot of temp sensors now are missed, now only GPU temp, TCTL and TDIE temp sensors. 
Before this version there were more sensors of the motherboard li.e. Voltage regulator.

This app also lacks the mechanical hard disk sensor temp.

Motherboard Gigabyte AB350N Gaming ITX, CPU Ryzen 7 1700, GPU RX 460 Yansolo B 2GB GDDR5.

corecoding, 2019-08-01:
Can you run the sensors command and share the output here?

mechano, 2019-08-01:
Here the output of the command.
I also upgraded the BIOS of my Gigabyte AB350N Gaming ITX to F41a.

mechano@ab350n:~$ sudo sensors
amdgpu-pci-0a00
Adapter: PCI adapter
vddgfx:       +0.85 V
fan1:        1546 RPM  (min =    0 RPM, max = 6000 RPM)
temp1:        +41.0°C  (crit = +94.0°C, hyst = -273.1°C)
power1:        9.24 W  (cap =  48.00 W)

k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +56.2°C  (high = +70.0°C)
Tctl:         +56.2°C

iwlwifi-virtual-0
Adapter: Virtual device
temp1:            N/A

corecoding, 2019-08-01:
Vitals doesn't know anything about computer hardware, so while your motherboard sounds nice, it doesn't help me. Vitals reads sensor data from the /proc and /sys directories as does the sensors command. These "files" (after all they aren't really directories) are data provided by the kernel.

Based on the sensors output above, which items are not showing in Vitals? It sounds like all of the sensors that are available are showing.

You could try running sensors-detect to see if the bios update botched some sensor modules on your system.

corecoding, 2019-08-10:
Closing due to no reply, let me know if you still need help.

closed 2019-08-10 by corecoding