corecoding/Vitals #102: Gnome Top Bar -  icon spacing changes on hover

matt-hires, 2019-09-12:
I would prefer this extension over my currently activated 'system-monitor'.
But when enabling the plugin in my gnome desktop the icon spacing in the top bar always changes on mouse hover:
[image]
*For reproduction try to restart gnome desktop and logout/login.*

**My System**
**Kernel:** 4.14.141-1-MANJARO x86_64 bits: 64 
**Desktop:** Gnome 3.32.2 Distro: Manjaro Linux

corecoding, 2020-01-25:
This can happen when one has too many icons in the menubar. That being said, I don't, and I've seen it since upgrading to the last two Gnome shell versions. Because no code was introduced, I truly believe this is a bug, possibly related to the ellipses bug. Not sure how to fix this one honestly.

corecoding, 2020-05-05:
Closing in favor of #141

closed 2020-05-05 by corecoding