corecoding/Vitals #103: Vitals stop working on Ubuntu 19.10 dev branch with gnome 3.34

foer, 2019-09-16:
Vitals stop working on Ubuntu 19.10/gnome 3.34. I can deliver debugging information if someone tells me how. In Looking Glass there is only the following error:

`TypeError: menuItems[i].key is undefined`

corecoding, 2019-09-16:
An update was just released today. Please confirm if the issue is fixed.

foer, 2019-09-17:
> Please confirm if the issue is fixed.

can't confirm that. The extension doesn't work after applying the update. The error message from above persists.

terencode, 2019-09-19:
Full log from master:
```
JS ERROR: Extension Vitals@CoreCoding.com: TypeError: menuItems[i].key is undefined
                                                   _appendMenuItem@/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:392:21
                                                   wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
                                                   _updateDisplay@/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:327:13
                                                   wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
                                                   _querySensors/<@/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:442:17
                                                   _returnValue@/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/sensors.js:406:9
                                                   wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
                                                   _queryStorage@/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/sensors.js:380:9
                                                   wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
                                                   query@/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/sensors.js:104:17
                                                   wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
                                                   _querySensors@/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:429:9
                                                   wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
                                                   _initializeTimer@/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:171:9
                                                   wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
                                                   _init@/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:85:9
                                                   wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
                                                   enable@/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:482:18
                                                   _callExtensionEnable@resource:///org/gnome/shell/ui/extensionSystem.js:131:13
                                                   loadExtension@resource:///org/gnome/shell/ui/extensionSystem.js:263:21
                                                   _loadExtensions/<@resource:///org/gnome/shell/ui/extensionSystem.js:473:13
                                                   collectFromDatadirs@resource:///org/gnome/shell/misc/fileUtils.js:27:17
                                                   _loadExtensions@resource:///org/gnome/shell/ui/extensionSystem.js:452:9
                                                   _enableAllExtensions@resource:///org/gnome/shell/ui/extensionSystem.js:482:13
                                                   _sessionUpdated@resource:///org/gnome/shell/ui/extensionSystem.js:513:13
                                                   init@resource:///org/gnome/shell/ui/extensionSystem.js:31:9
                                                   _initializeUI@resource:///org/gnome/shell/ui/main.js:238:5
                                                   start@resource:///org/gnome/shell/ui/main.js:138:5
                                                   @<main>:1:31
sept. 19 22:54:07 terence-arch gnome-shell[37335]: JS WARNING: [/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js 392]: reference to undefined property "key"
sept. 19 22:54:07 terence-arch gnome-shell[37335]: Usage of object.actor is deprecated for VitalsMenuButton
                                                   get@resource:///org/gnome/shell/ui/environment.js:240:29
                                                   _init@/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:69:9
                                                   wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
                                                   enable@/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:482:18
                                                   _callExtensionEnable@resource:///org/gnome/shell/ui/extensionSystem.js:131:13
                                                   loadExtension@resource:///org/gnome/shell/ui/extensionSystem.js:263:21
                                                   _loadExtensions/<@resource:///org/gnome/shell/ui/extensionSystem.js:473:13
                                                   collectFromDatadirs@resource:///org/gnome/shell/misc/fileUtils.js:27:17
                                                   _loadExtensions@resource:///org/gnome/shell/ui/extensionSystem.js:452:9
                                                   _enableAllExtensions@resource:///org/gnome/shell/ui/extensionSystem.js:482:13
                                                   _sessionUpdated@resource:///org/gnome/shell/ui/extensionSystem.js:513:13
                                                   init@resource:///org/gnome/shell/ui/extensionSystem.js:31:9
                                                   _initializeUI@resource:///org/gnome/shell/ui/main.js:238:5
                                                   start@resource:///org/gnome/shell/ui/main.js:138:5
                                                   @<main>:1:31
```

corecoding, 2019-09-23:
I’m looking into it. Please know that this extension is not marked with 3.34 compatibility. Also, I’m accepting pull requests if anyone can help expedite the process.

corecoding, 2019-09-28:
Can you guys take a look at the develop branch and let me know if it works for you? There are a couple deprecation warnings but it is loading for me now.

\

fikrimi23, 2019-09-28:
> Can you guys take a look at the develop branch and let me know if it works for you? There are a couple deprecation warnings but it is loading for me now.
> 
> \

develop branch is working, Gnome 3.34.0, with this warning 
```
Usage of object.actor is deprecated for VitalsMenuButton
get@resource:///org/gnome/shell/ui/environment.js:242:29
_init@/home/fikrimi/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:69:9
wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
enable@/home/fikrimi/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:463:18
_callExtensionEnable@resource:///org/gnome/shell/ui/extensionSystem.js:131:13
_onEnabledExtensionsChanged/<@resource:///org/gnome/shell/ui/extensionSystem.js:398:13
_onEnabledExtensionsChanged@resource:///org/gnome/shell/ui/extensionSystem.js:395:9

```

foer, 2019-09-28:
It is working now under ubuntu 19.10/Ggnome 3.34. Thank you!

kouros17, 2019-10-01:
Could you add it to master branch please?

corecoding, 2019-10-14:
It is in the master branch and also currently released in gnome extensions. :)

closed 2019-10-14 by corecoding