corecoding/Vitals #104: Wrong icons and formatting for Network Monitor part

UndeadDevel, 2019-09-17:
What's wrong?
https://postimg.cc/qNY9tVRq
In the above partial screenshot you see the extension with CPU average, Memory usage and Network down / up enabled. I opened the Network section and you see the wrong formatting (unnecessary insets and unnecessarily large line spacing that the other sections don't have) as well as wrong icons for the Network data (shows the process explorer thing icon). This used to work properly but broke the update before last, so if this is now version 22 then I guess it broke in version 20; before that it worked fine.
Also, Fan and Voltage have never worked for me with this extension; it always displays "No Data".

I'm on Fedora 30 with Gnome 3.32.2.

What have you tried?
disabling, re-enabling the extension; switching off network monitoring, re-enabling it...I hoped the next update would fix it but it's still broken.
With regards to Fan and Voltage I tried "hide zero values" option but that didn't change anything.

Automatically detected errors:

GNOME Shell Extensions did not detect any errors with this extension.

Version information:

    Shell version: 3.32.2
    Extension version: 22

SmartFinn, 2019-09-17:
That issue related to shared-mime-info v1.13.0/v1.13.1 See [BUG: Invisible icons in 1.13 version](https://gitlab.freedesktop.org/xdg/shared-mime-info/issues/116).

corecoding, 2019-09-23:
Interesting. Those network icons are included so not sure why they are missing. This is the first I’ve heard of this issue. In order to fix I have to duplicate it. I’ll need the exact os you have installed, with the exact packages / extensions you installed so I can see it first hand.

SmartFinn, 2019-09-23:
@corecoding it's reproduced on any disto (ex. Fedora 30) with shared-mime-info v1.13.0/v1.13.1 installed.

It also can be fixed on your side, just formating the following files: network-download-symbolic.svg, network-symbolic.svg, and network-upload-symbolic.svg with svgo/scort/svgcleaner. Or wait when shared-mime-info v1.14 will be packaged for those distros.

UndeadDevel, 2019-09-23:
OK...after preparing a huge post with all sorts of system info I see that @SmartFinn says that share-mime-info v1.14 fixes the issue so I check Fedora updates testing repo and it has 1.14, which I update to and voila it works haha...
So the solution is to upgrade shared-mime-info to version 1.14.

Any idea why Voltage and Fan are showing "No Data", though? Those still don't work.

dac73, 2019-10-17:
For me "No Data" shows for Fan when RPM is 0, but it doesn't change when it starts spinning unless I restart the extension.
So it looks like that extension stops tracking the sensor once it reports 0 RPM.

corecoding, 2020-05-05:
Glad to hear the network monitor is working. Voltage may not be shown because it may not be available, or it is a bug we have fixed a while ago. For fan, there is the "Hide zero values" option that you can now turn off.

closed 2020-05-05 by corecoding