corecoding/Vitals #108: Doesn't work with gnome 3.34.1 on Manjaro

kouros17, 2019-10-08:
I installed it from extensions.gnome.org (3.34) but not loaded at all (of course I restarted gnome-shell and enabled it from gnome-tweak-tool).

robvelor, 2019-10-09:
Same here, doesn't work on archlinux on 3.34.

kouros17, 2019-10-09:
I installed it from git (with fix for #108) but doesn't work also.

```State change on user/*/*/shell-extension/Vitals_CoreCoding.com/* from available to installed is not OK ```

```
Usage of object.actor is deprecated for VitalsMenuButton
get@resource:///org/gnome/shell/ui/environment.js:242:29
_init@/home/giorgos/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:70:13
wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
enable@/home/giorgos/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:467:18
_callExtensionEnable@resource:///org/gnome/shell/ui/extensionSystem.js:131:13
_onEnabledExtensionsChanged/<@resource:///org/gnome/shell/ui/extensionSystem.js:398:13
_onEnabledExtensionsChanged@resource:///org/gnome/shell/ui/extensionSystem.js:395:9
```

corecoding, 2019-10-09:
Try from gnome extension store. Fix was on master branch. 

Christopher Monahan

robvelor, 2019-10-09:
Last update on the extension store fixed it for me, thx!

kouros17, 2019-10-09:
Unfortunately it doesn't work

corecoding, 2019-10-09:
At this point I don’t have necessary information to duplicate your issue. As mentioned, feel free to create an image of Manjaro with the 3.34.1 and vitals installed for Gnome Boxes. If I can duplicate what you are seeing, I can fix it.

kouros17, 2019-10-11:
Finally I have the same issue like [this](https://github.com/corecoding/Vitals/issues/109).
The add-on is there just doesn't show its icon on the panel. I had to choose some values first.

closed 2019-10-11 by kouros17