corecoding/Vitals #109: No displaying on panel

abriel, 2019-10-10:
Hi,

I completely don't see the monitoring on my panel. 
[image]

Just with luck I found a black place on the panel, and able to see the app menu.

Please advice.

Arch linux, gnome 3.34.
Extension from extensions.gnome.org , for shell 3.34, version 24.

Thanks,

kouros17, 2019-10-11:
I have the same issue on Manjaro. The add-on is there just doesn't show its icon on the panel.You have to choose some values first.

abriel, 2019-10-11:
@kouros17 , 
Oh my god!!! I spent about a week trying to cleanup configuration and re-installing the add-on. Nothing did help. But... Eventually it is so simple!
I just did miss the moment when all metrics gone from the panel.
Ok, I re-add them, now I got it.. So much thanks to you!

How do you think, should we close this?

kouros17, 2019-10-11:
> Oh my god!!! I spent about a week trying to cleanup configuration and re-installing the add-on. Nothing did help.
Μe too. :)

corecoding, 2019-10-14:
Glad you figured it out. There is still an issue, perhaps. If all sensors are unchecked, there should be a default icon that displays so you have something to click on. The icon I use is an operating system provided icon, so perhaps gnome-icon-theme-symbolic is not installed. Can you check?

abriel, 2019-10-14:
Hi @corecoding , here it is:
```
$ yaourt -Ss gnome-icon-theme-symbolic
extra/gnome-icon-theme-symbolic 3.12.0-5 [installed]
    GNOME icon theme, symbolic icons
```
I use Adwaita theme, which is default.

Thanks.

corecoding, 2020-05-05:
Closing because I have not heard any other reports of this.

closed 2020-05-05 by corecoding