corecoding/Vitals #118: Public IP Address not updating

wolfen351, 2019-11-03:
If I connect to a VPN after the session starts, then the public IP shown is not updated to reflect the new one

Steps to reproduce the behavior:
1. Login to a fresh session
2. Wait for everything to load
3. Observe the public IP in the top bar (A)
4. Connect to a VPN that routes all traffic over the VPN
5. Open terminal and run "curl icanhazip.com" - Observe the output (B)
6. Observe the top bar ip address is still (A) and not (B)

**Expected behavior**
I would expect that once every couple of mins that the public IP address is refreshed.

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04.3 LTS (Bionic Beaver)
 - Gnome Version (3.28.2)

**Additional context**
It does update if you remove and then re-add the IP to the top bar.

corecoding, 2019-11-03:
The IP is the only sensor that doesn’t adhere to the update interval in preferences. I think I set it up to check every 5 minutes. You should also be able to press the refresh button and have it update right away. 

Christopher Monahan

wolfen351, 2019-11-03:
Thanks for the response. I had a quick look at the code and it seems it is set up for every 900 seconds - so approx every 15 mins. Its great to know that I can manually refresh it, thanks!

wolfen351, 2019-11-04:
I would love the following to be implemented:

1) Get all local IP addresses, save them in an internal array
2) Wait 30 seconds
3) Get all local IP addresses, save them in a second internal array
4) Compare the 2 arrays, if they differ, then do the public ip check
5) Update the initial internal array with the new local ip addresses
6) Go to 2

This should catch many of the "network change" events, like connecting to wifi/hotspot/ethernet/vpn etc

shanmukhateja, 2020-05-16:
NetworkManager has a DBus API with VPN support. So, if we listen to it, we could trigger function to fetch the public IP manually.

This way it'll be event driven instead of polling.

Here's the link:

[Link](https://developer.gnome.org/NetworkManager/stable/nm-vpn-dbus-types.html

The caveat is, this might need root access. I'm noy entirely sure however, this seems like a start!