corecoding/Vitals #119: Units are ellipsized after a shell restart

terencode, 2019-11-23:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 19.04]
 - Gnome Version [e.g. 3.30]

**Additional context**
Add any other context about the problem here.

terencode, 2019-11-24:
With https://github.com/corecoding/Vitals/commit/53cb50c7889766c182c1027a5da99e104284c81f it sometimes works but not always.

corecoding, 2019-11-24:
I contacted a gnome shell developer for this problem a while back. With 3.34, parts of gnome shell aren't updated /  refreshed as often to give it the speed we have grown to love. He suggested this.emit    this.emit('style-changed');   but didn't say where to put it. I just added it to the enable/disable area. Maybe it's fixed? Finger crossed.

corecoding, 2019-11-24:
Never mind, this.emit isn't working. Anyhow, I think the issue now is when my laptop goes to external monitor (hi to lo dpi).. I notice other weird things (not related to Vitals) such as some buttons

nebrelbug, 2020-03-23:
@corecoding any updates? Loving this extension but units are ellipsized every time I open up my computer :man_shrugging: :joy:

corecoding, 2020-05-11:
With the recent changes I have made, I have seen the ellipses less and less. It's tough because there is no doubt something I can change in the code to prevent them, but I have no idea what. They started happening somewhere around Gnome 3.30 - without any changes in Vitals code. When the next version of Vitals is released, please let me know if they go away.

terencode, 2020-05-11:
Yes, I don't see anything hidden right now on version 32 (is it a mistake? last version was 35: https://github.com/corecoding/Vitals/commit/38f8bb690b1c1d78538709b02a166b46f0a4dfc6)

corecoding, 2020-05-11:
I had set version 35 so Ubuntu didn't update the extension with the the live version, erasing any uncommitted development changes. 35 was just me taking a guess at the next version number. 32 is the official next version number.

blipk, 2020-05-29:
@corecoding you can try `St.Label.get_clutter_text().ellipsize = 0;` on the 'HotItem' labels

corecoding, 2020-05-29:
Thank you for this tip. Added to develop branch. Let me know if it helps!

corecoding, 2020-06-04:
Was anyone able to confirm this fix? (On develop branch). I think it is working!

terencode, 2020-06-04:
I can't speak for myself because I couldn't already reproduce before but maybe it has to do with not using gdm lockscreen anymore...
However, switching to develop branch I notice big spacing with the first and last elements: 
[image]

nebrelbug, 2020-06-05:
@corecoding may be worth noting that for some reason, I (on GNOME 3.36) only notice the RAM% as being ellipsized, not core temperature or CPU%

terencode, 2021-01-15:
Wondering if we should close this, I personally don't get the problem anymore.

corecoding, 2021-03-18:
I haven't seen it in a long while either. Closing for now.

closed 2021-03-18 by corecoding

wmertens, 2022-05-24:
I'm seeing this. Gnome 42, was also happening in 41. When I freshly added the extension, it was fine. Now, it's like this 100% of the time.

None of the display settings make any difference. There's plenty of room on the top bar. On/off extension also makes no difference.

[image]

wmertens, 2022-05-24:
oh wow - it turns out those ... are metrics that were no longer available. As soon as I selected one they were gone. Problem solved!