corecoding/Vitals #124: Support AMD GPU metrics via sysfs

whentze, 2020-01-19:
**Describe the new feature you'd like**
It would be useful if the extension displayed GPU load and memory usage for AMD GPUs.

**Additional context**

I know this gets requested a lot, but in the case of new-ish AMD GPUs (i.e. ones supported by the `amdgpu` open source driver), it is much easier to support.
The driver exposes at least the following values via sysfs:

- `/sys/class/drm/cardN/device/gpu_busy_percent` : current GPU load in %
- `/sys/class/drm/cardN/device/pp_dpm_sclk` : GPU clock rates available (current clock rate on the line with a "*")
- `/sys/class/drm/cardN/device/mem_info_vram_used` : VRAM usage in bytes
- `/sys/class/drm/cardN/device/mem_info_vram_total` : total VRAM in bytes

corecoding, 2020-01-25:
I think this falls in line with what Vitals is, and as mentioned, that is a non io blocking extension. I think it makes sense to put these in into their own section called Graphics. Assuming cardN is a number, you could iterate from 0 to say 5, and use that number in the human readable text.

Unfortunately, not having this graphics card makes it very hard for me to see if what I coded is working. If you can make the changes and create a PR, I'd be glad to merge it in!

eddie-dunn, 2022-10-14:
@corecoding I have an AMD card and could look into it. I know JS/TS like Neo knows kung fu, but I know nothing about gnome extensions.

Can you give me some pointers on where in this code base I should start looking when implementing this feature?

renattele, 2024-03-16:
[Resources](https://flathub.org/apps/net.nokyan.Resources) can monitor GPU stats: [source](https://github.com/nokyan/resources/blob/main/src/utils/gpu/amd.rs), their implementation works for me(integrated AMD GPU + discrete AMD RX GPU). We can use this to implement stats.

velias, 2024-11-29:
PR for this feature is https://github.com/corecoding/Vitals/pull/456

corecoding, 2026-09-05:
feature released

closed 2026-09-05 by corecoding