corecoding/Vitals #128: Code updates

oleastre, 2020-03-05:
Hello, there have been some new updates of your extension on ego (probably related to new translations), but there is no code change in this repository since December.
Is it just an oversight ?

corecoding, 2020-03-05:
It is not an oversight. I never intended Vitals to be hugely feature rich as I don’t like daemons and widgets that hog resources. At the time, I was running a lower resource Linux laptop and needed to monitor these things. I am still here to accept PRs from the community.

oleastre, 2020-03-09:
I have the feeling that my question was not understood correctly.
What I meant is that the version distributed on extensions.gnome.org is at version 27, and here the latest version is 25. 
So my question is simply: did you forget to push some changes  ? 
(and from a diff between the two, I think it's mainly related to translations)

corecoding, 2020-03-09:
Oh! I see what you are asking. Master branch here is the same as the version on gnome extension website. Sometimes I get lazy and forget to increment the version number, which is done automatically every time you upload a new release.

oleastre, 2020-03-10:
I just did a quick diff and indeed, except the added Russian translation and version number, there are no change. 
Maybe just add the Russian translation to this repo and we can close the issue.

closed 2020-03-10 by corecoding

oleastre, 2020-03-11:
Thanks for the sync and for your work on this extension.