corecoding/Vitals #129: error under GNOME 3.36

coxackie, 2020-03-08:
updates to GNOME 3.36 rolled out through arch linux - extension shows an error.

SamboyCoding, 2020-03-08:
On Ubuntu 20.04 this errors too - same GNOME update. The error is `TypeError: panelSystem._createActionButton is not a function` which presumably refers to this line (as well as the two further calls to this function below it): https://github.com/corecoding/Vitals/blob/2aa5cb41e8a13c49110da4acefb387255a6088a6/extension.js#L130

From using the new shell what I can surmise is that the buttons that were being used at the bottom of the vitals list no longer exist - they were used for the power, and settings icons in the system menu, which is what this code is attempting to use, but those options in the system menu now look like this: 
[image]

So if I had to guess, I'd say the function doesn't exist anywhere anymore, and those buttons will have to be changed or entirely reimplemented.

Finally, here is the commit in which the function was removed from the main gnome-shell repository - which explains their reasoning somewhat: https://github.com/GNOME/gnome-shell/commit/147a743d8d7947d99e274861a34403aaa204324e#diff-d6e0e57ec48960f580ed780b11890e7a

kouros17, 2020-03-09:
On manjaro doesn't work:

```
JS ERROR: Extension Vitals@CoreCoding.com: TypeError: panelSystem._createActionButton is not a function
_initializeMenu@/home/giorgos/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:130:39
wrapper@resource:///org/gnome/gjs/modules/script/_legacy.js:82:27
_init@/home/giorgos/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:96:14
wrapper@resource:///org/gnome/gjs/modules/script/_legacy.js:82:27
enable@/home/giorgos/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:479:18
_callExtensionEnable@resource:///org/gnome/shell/ui/extensionSystem.js:160:32
loadExtension@resource:///org/gnome/shell/ui/extensionSystem.js:311:26
_loadExtensions/<@resource:///org/gnome/shell/ui/extensionSystem.js:536:18
collectFromDatadirs@resource:///org/gnome/shell/misc/fileUtils.js:27:17
_loadExtensions@resource:///org/gnome/shell/ui/extensionSystem.js:515:19
_enableAllExtensions@resource:///org/gnome/shell/ui/extensionSystem.js:545:18
_sessionUpdated@resource:///org/gnome/shell/ui/extensionSystem.js:576:18
init@resource:///org/gnome/shell/ui/extensionSystem.js:54:14
_initializeUI@resource:///org/gnome/shell/ui/main.js:244:22
start@resource:///org/gnome/shell/ui/main.js:138:5
@<main>:1:47
```
And:

```
Some code tried to set a deprecated GObject property.
0 _patchContainerClass/containerClass.prototype.child_set() ["resource:///org/gnome/shell/ui/environment.js":32:12]
1 _patchContainerClass/containerClass.prototype.add() ["resource:///org/gnome/shell/ui/environment.js":41:17]
2 _createHotItem() ["/home/giorgos/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js":198:25]
3 wrapper() ["resource:///org/gnome/gjs/modules/script/_legacy.js":82:26]
4 _drawMenu() ["/home/giorgos/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js":283:17]
5 wrapper() ["resource:///org/gnome/gjs/modules/script/_legacy.js":82:26]
6 _init() ["/home/giorgos/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js":75:13]
7 wrapper() ["resource:///org/gnome/gjs/modules/script/_legacy.js":82:26]
8 enable() ["/home/giorgos/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js":479:17]
9 _callExtensionEnable() ["resource:///org/gnome/shell/ui/extensionSystem.js":160:31]
10 loadExtension() ["resource:///org/gnome/shell/ui/extensionSystem.js":311:25]
11 _loadExtensions/<() ["resource:///org/gnome/shell/ui/extensionSystem.js":536:17]
12 collectFromDatadirs() ["resource:///org/gnome/shell/misc/fileUtils.js":27:16]
13 _loadExtensions() ["resource:///org/gnome/shell/ui/extensionSystem.js":515:18]
14 _enableAllExtensions() ["resource:///org/gnome/shell/ui/extensionSystem.js":545:17]
15 _sessionUpdated() ["resource:///org/gnome/shell/ui/extensionSystem.js":576:17]
16 init() ["resource:///org/gnome/shell/ui/extensionSystem.js":54:13]
17 _initializeUI() ["resource:///org/gnome/shell/ui/main.js":244:21]
18 start() ["resource:///org/gnome/shell/ui/main.js":138:4]
19 <TOP LEVEL> ["<main>":1:46]
```

SamboyCoding, 2020-03-09:
Didn't get the second one personally but the first one is the one that's breaking everything. I do sorta feel like this repo is abandoned so we'll probably have to wait for some kind soul to fork.

corecoding, 2020-03-09:
The repo is not abandoned. Will review and merge when I get back on a computer, which will be tomorrow. Please note that with arch, you are cutting edge but patience is required to make sure vitals works for all previous gnome shells as well.

SamboyCoding, 2020-03-09:
Now that's a pleasant surprise! 😄 One of my favourite extensions

kouros17, 2020-03-12:
I can't live without it. Please upgrade. :)

coxackie, 2020-03-18:
@corecoding are there any updates on using vitals under 3.36?

corecoding, 2020-03-19:
Still waiting for a pull request from the community. My previous post was made assuming that’s what this issue was about, but it was just a link to the gnome code base. I have to be honest, I am not happy in how Gnome has been handling the last 4 releases. But I digress.

befedo, 2020-03-20:
> Still waiting for a pull request from the community. My previous post was made assuming that’s what this issue was about, but it was just a link to the gnome code base. I have to be honest, I am not happy in how Gnome has been handling the last 4 releases. But I digress.

Don't worry! You've done an amazing work with this extension and many ... many users where out there in the community. I think someone attends to this issue in the next time ...

Greatings and Heads up! :thumbsup:

corecoding, 2020-03-22:
I have done some work in the develop branch, if you want to check it out. It doesn't crash anymore, however, the preferences page still doesn't load. I can't find any extension on extensions.gnome.org that has a preferences page that loads. Perhaps it is because I am running Ubuntu 20.04?

dasnoopy, 2020-03-22:
..tested  **dev** branch here on **archlinux** and **gnome 3.36 :**

there is a weird code line in the file `extensions.js:`

`JS ERROR: Extension VitalsDev@CoreCoding.com: SyntaxError: expected expression, got '>>>' @ /home/user/.local/share/gnome-shell/extensions/VitalsDev@CoreCoding.com/extension.js:182
`

After I removed that line, extension start and seems it works properly. Also the preferences page load and works properly, here.

thanks and regards
Andrea

fikrimi23, 2020-03-22:
currently working, for line 182, is specifically this line 
https://github.com/corecoding/Vitals/blob/57635f3377bdab17ac2061571779247e95f81c92/extension.js#L182

Eskander, 2020-03-22:
> currently the extension working properly and for me, the extensions preferences is also working,
> for line 182, is specifically this line
> 
> https://github.com/corecoding/Vitals/blob/57635f3377bdab17ac2061571779247e95f81c92/extension.js#L182

Can confirm. Removing this line from `develop` branch makes Vitals work in GNOME 3.36 on Arch.

kouros174, 2020-03-22:
I confirm it too in manjaro. Removing this line one of my favorite extensions is back again (and the preferences as well). :+1:

cantalupo555, 2020-03-22:
> The repo is not abandoned. Will review and merge when I get back on a computer, which will be tomorrow. Please note that with arch, you are cutting edge but patience is required to make sure vitals works for all previous gnome shells as well.

thank you

corecoding, 2020-03-22:
Good to hear Preferences are working. Must have been an issue with Ubuntu 20.04 - after all, it is still beta. I pushed the fix to remove that line. It was due to a merge conflict that I didn't test.

SamboyCoding, 2020-03-24:
Can confirm I'm also on ubuntu 20.04 and preferences don't work for any extension for me - rather annoying though that is.

dukase, 2020-03-24:
Try install extension preference module which is missing in 20.04.
_sudo apt install gnome-shell-extension-prefs_

bradleyhop, 2020-03-25:
The develop branch solves this issue for me as well: works and preferences are displayable. Manjaro with 3.36.0 gnome.

dukase, 2020-03-25:
How can you get the develop branch?

linuxiorr, 2020-03-26:
The develop branch works for me as well (preferences work too). I'm using Solus with GNOME 3.36.0

corecoding, 2020-03-26:
master updated, and new version submitted to Gnome Extensions. Could take an hour, could take weeks. Depends on their approval time. You can see it in the queue here:
https://extensions.gnome.org/review/

terencode, 2020-03-27:
Nice, still getting this warning though:
```
Some code tried to set a deprecated GObject property.
                                                    0 _patchContainerClass/containerClass.prototype.child_set() ["resource:///org/gnome/shell/ui/environment.js":32:12]
                                                    1 _patchContainerClass/containerClass.prototype.add() ["resource:///org/gnome/shell/ui/environment.js":41:17]
                                                    2 _createHotItem() ["/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js":228:25]
                                                    3 wrapper() ["resource:///org/gnome/gjs/modules/script/_legacy.js":82:26]
                                                    4 _drawMenu() ["/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js":313:17]
                                                    5 wrapper() ["resource:///org/gnome/gjs/modules/script/_legacy.js":82:26]
                                                    6 _init() ["/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js":75:13]
                                                    7 wrapper() ["resource:///org/gnome/gjs/modules/script/_legacy.js":82:26]
                                                    8 enable() ["/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js":509:17]
                                                    9 _callExtensionEnable() ["resource:///org/gnome/shell/ui/extensionSystem.js":160:31]
                                                    10 loadExtension() ["resource:///org/gnome/shell/ui/extensionSystem.js":311:25]
                                                    11 _loadExtensions/<() ["resource:///org/gnome/shell/ui/extensionSystem.js":541:17]
                                                    12 collectFromDatadirs() ["resource:///org/gnome/shell/misc/fileUtils.js":27:16]
                                                    13 _loadExtensions() ["resource:///org/gnome/shell/ui/extensionSystem.js":520:18]
                                                    14 _enableAllExtensions() ["resource:///org/gnome/shell/ui/extensionSystem.js":550:17]
                                                    15 _sessionUpdated() ["resource:///org/gnome/shell/ui/extensionSystem.js":581:17]
                                                    16 init() ["resource:///org/gnome/shell/ui/extensionSystem.js":54:13]
                                                    17 _initializeUI() ["resource:///org/gnome/shell/ui/main.js":249:21]
                                                    18 start() ["resource:///org/gnome/shell/ui/main.js":138:4]
                                                    19 <TOP LEVEL> ["<main>":1:46]

```

corecoding, 2020-04-01:
Sigh. Unfortunately my Linux box died, so all I have is my Mac. I will see if I can get a Raspberry Pi 4 B or something like that for easy development. This warning will take a bit to get to.

terencode, 2020-04-01:
Oh rip

corecoding, 2020-05-02:
I am going to close this issue. @terencode - If you are still seeing this, can you open another issue? I am not seeing it. Thanks everyone.

closed 2020-05-02 by corecoding

terencode, 2020-05-02:
Thanks @corecoding, I can still reproduce so here it is: #140