corecoding/Vitals #131: CPU reporting wrong percent

battata, 2020-03-29:
CPU% reporting is wrong
ie:
[image]

czux, 2020-03-29:
I'm having the same issue.
[image]

corecoding, 2020-04-13:
Thank you for your bug report. Development is paused until my goal is met. Please see https://github.com/corecoding/Vitals/issues/134 for more info.

corecoding, 2020-05-02:
I have a few questions for you!

1. How are you maxing out your cpu? I am using the stress command. 
2. When your not maxed out, does the percentage in Vitals work correctly?
3. Can you attach a snapshot of /proc/stat when your cpu is maxed out?

corecoding, 2020-05-05:
Note to self - sometimes CPU higher than 100%

battata, 2020-05-06:
1. I'm maxing out the CPU with folding@home. Sorry I couldn't respond earlier, I upgraded to Ubuntu 20.04 and had trouble installing the folding client.
2. I don't know, I assume so, the percents are too little.
3. 
> cpu  1987674 625818 577846 65366970 43697 0 48670 0 0 0
cpu0 254539 73146 67787 8174224 5135 0 18069 0 0 0
cpu1 253027 81692 66468 8176502 4783 0 7950 0 0 0
cpu2 254576 77726 64131 8184730 3086 0 3780 0 0 0
cpu3 258885 60824 72754 8183538 3779 0 1373 0 0 0
cpu4 249638 97197 66230 8169183 2820 0 890 0 0 0
cpu5 219676 62381 103966 8177596 3416 0 566 0 0 0
cpu6 243689 87118 67634 8171453 14497 0 1173 0 0 0
cpu7 253640 85731 68873 8129741 6177 0 14865 0 0 0
intr 112102979 6 4 0 0 0 0 0 0 1 0 0 0 0 0 0 0 0 737 378109 0 0 0 0 31 0 0 0 0 0 19955019 932536 17 1249 6051710 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
ctxt 349921234
btime 1588715669
processes 148790
procs_running 8
procs_blocked 0
softirq 111252148 378008 36617415 598709 19974547 924987 0 222078 38813382 152060 13570962

corecoding, 2020-05-09:
Can you try the develop branch and see if the issue is fixed?

battata, 2020-05-09:
Tested and working as intended.
Thank you!

closed 2020-05-09 by battata