corecoding/Vitals #133: Drop down menu not centered

shbach, 2020-04-02:
**Describe the bug**
A clear and concise description of what the bug is.

[image]

Currently it seems like the drop down menu is centered based off the left side of the menu.

**Expected behavior**
The dropdown menu should be in the very center of the icons. Everything else in the top bar has their dropdown menu centered so this doesn't match.

**Desktop (please complete the following information):**
 - OS: Solus 4.1
 - Gnome Version 3.36

shbach, 2020-04-04:
Actually, this seems to be an issue with more than one extension so it might be a problem with gnome extensions in general.

vesterlay, 2020-04-09:
I have the same issue. It's caused by vitals, you can repair tray by disabling vitals and restarting gnome shell(alt+f2 -> r).

corecoding, 2020-04-13:
Thank you for your bug report. Development is paused until my goal is met. Please see https://github.com/corecoding/Vitals/issues/134 for more info.

corecoding, 2020-05-03:
Thanks for your bug report. The menu actually where I want it. Take a look at my screenshot, you are missing an arrow which may make my decision a little more apparent.
[image]

closed 2020-05-03 by corecoding