corecoding/Vitals #134: Community Help

corecoding, 2020-04-07:
Hello everyone,

Thank you for making Vitals a part of your desktop experience. Its success has surprised me; it has seen around 300k installations since mid 2018. I released Vitals with the goal of paying back the open source community. I have a small donation button in the help screen (and here on GitHub) but that's it. Making it nagware would go against the very essence of what open source is and I am having fun maintaining it. That being said, I was surprised to see how few actually donated. I've received about $30. I've only donated to a few open source projects through the years, so I get definitely get it.

That being said, I'm running a Mac full time for my other projects. My last Linux box died, so I am asking for donations to buy a $99 Raspberry Pi 4 4GB to continue development. Unfortunately, I can't commit to any features but this will definitely help the project go on longer.

Just wanted to throw this out there. Hope everyone is staying healthy.

WoefulDerelict, 2020-04-07:
I've echoed this over in the AUR. Hopefully some Arch users who enjoy the extension and use the PKGBUILD will step up.

corecoding, 2020-04-12:
$89.32 left. :)

corecoding, 2020-04-13:
$88.66 left

vesterlay, 2020-04-15:
Please, add donation link to readme, I can't find it anywhere on the site.

corecoding, 2020-04-15:
Sure thing! Thank you.

corecoding, 2020-04-15:
$84.54 left - thank you everyone!

corecoding, 2020-04-17:
$77.19 left 👍

esa1975, 2020-04-28:
Hopefully I can get some awareness out there. I don't have a huge following but every little bit helps I suppose!

https://youtu.be/dYKfV_YYedk

corecoding, 2020-04-30:
Thank you for creating this video. Yesterday was a tough day at work and this made my day. I also received some donations. Looking forward to addressing some of the community requests!

corecoding, 2020-04-30:
Just wanted to give another update, two people came in and donated (in my opinion) more than I would have expected any single person to donate. If you are curious, the balance is now at $43.81. I honestly have felt weird even asking, as I understand open source is about contributing, not making money. Thankfully, I have procured an old Dell Desktop, and will be using this to continue the development. It doesn't support my screen resolution very well, so I used the donations to purchase a used video card off eBay that will run my monitor.

Again thank you for the support, and especially the YouTube video. Pretty neat stuff!

esa1975, 2020-05-01:
I'm very glad to hear you received some donations. Thanks again for your work on this. I appreciate having such a well designed piece of software available. Isn't open source great?

the-mentor, 2020-05-01:
@corecoding what's your PayPal?

shanmukhateja, 2020-05-01:
Hi!

I have sent $25 via PayPal. Thank you for Vitals extension :)

joephill11, 2020-05-01:
Just donated 20$

KillYourFM, 2020-05-01:
@the-mentor @joephill11 @shanmukhateja You guys are heroes.

popey, 2020-05-01:
Never used your extension before. Sounds neat though, so lobbed $20 in. Keep up the good work.

corecoding, 2020-05-02:
Just wanted to say thanks again everyone. I woke up yesterday to my phone blowing up, and found out that @esa1975  had contacted @KillYourFM about my plea for help. I went from feeling burned out (amplified by lack of a Linux box) to feeling rejuvenated and quite honestly, a little guilty for even asking for help. The community stepped up in a major way. Instead of a Raspberry Pi, I ended up ordering a Pinebook Pro, as someone suggested on Twitter. I think this machine will work quite well for continuing development! Before it is shipped, can anyone confirm that Gnome works? It doesn't matter, but it appears Fedora recently added pinebook / Gnome support. Just want to make sure.

the-mentor, 2020-05-02:
@corecoding i have a friend that has the pinebook Pro and he says gnome works on it.
He also said it can be a bit laggy but usable.

corecoding, 2020-05-10:
@the-mentor thanks! I decided to cancel my order, and went a different direction. :)

closed 2020-05-10 by corecoding