corecoding/Vitals #135: battery percentage false

x0rld, 2020-04-12:
**Describe the bug**
I tried to use the battery monitoring but it always stays at 100%

**To Reproduce**
Steps to reproduce the behavior:
1 activate the battery monitoring
2 look at the battery capacity percentage
**Expected behavior**
the battery capacity percentage always stays at 99%
**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [fedora 31
 - GNOME Shell 3.34.5

corecoding, 2020-04-13:
Thank you for your bug report. Development is paused until my goal is met. Please see https://github.com/corecoding/Vitals/issues/134 for more info.

corecoding, 2020-05-05:
Can you tell me which, if any, of the directories you have files in? 

 /sys/class/power_supply/BAT0
 /sys/class/power_supply/BAT1
 /sys/class/power_supply/BAT2 
etc...

x0rld, 2020-05-05:
i have BAT0 with think symlink
`/sys/class/power_supply/BAT0 -> 
../../devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0/
`

x0rld, 2020-05-08:
the file capacity has the good value

corecoding, 2020-05-08:
Can you share what is in the file?

x0rld, 2020-05-08:
sure 
[image]

corecoding, 2020-05-09:
Thanks! re the rest of the battery stats correct? Can you send me a screenshot of the battery section UI from Vitals?

x0rld, 2020-05-10:
all the datas are here, I didn't find all the datas in /sys/class/power_supply/BAT0/capacity
[image]

corecoding, 2020-05-10:
Glad it is working. I should have realized this before, but Capacity is your battery health. 99% is good, meaning your battery has only degraded 1%. The Percentage is your battery level.

closed 2020-05-10 by corecoding