corecoding/Vitals #136: Vitals cause topbar glitching after locking the screen in 3.36

Grizzlywolf, 2020-04-13:
This extensions, when enabled, causes weird topbar status icon glitching when hovering over them after locking the screen. The only way to fix the issue after that is to restart the shell. Here is a video that shows it in action: [https://i.imgur.com/AKgKyZu.mp4](https://i.imgur.com/AKgKyZu.mp4)

corecoding, 2020-04-13:
Thank you for your bug report. Development is paused until my goal is met. Please see https://github.com/corecoding/Vitals/issues/134 for more info.

corecoding, 2020-05-02:
I have seen this before, but not for a long while. Can you disable your other extensions, one by one until it stops doing this? Then, let me know which extension is causing it. I can install that extension and see how it interacts with Vitals.

corecoding, 2020-05-05:
Closing in favor of #141

closed 2020-05-05 by corecoding