corecoding/Vitals #139: Storage shows "No data"

popey, 2020-05-02:
**Describe the bug**
The storage category has "No data". I have followed the steps in https://github.com/corecoding/Vitals/wiki/Storage-has-no-items and seen issue #114 and this doesn't fix it. 

A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Ubnutu 20.04 (on experimental ZFS support - possibly why?)
2. Install Vitals
3. Follow guide on https://github.com/corecoding/Vitals/wiki to add additional packages
4. Open Vitals, see "No data" next to storage

**Expected behavior**
I dunno what should appear there, as this is my first install of Vitals, but I expect something which suggests free space on the root filesystem.
**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - Gnome Version 3.36.1

**Additional context**
Possibly the fact I'm using ZFS which is new in 19.10, may be why?

corecoding, 2020-05-02:
Thanks for the donation today! Can you try logging out and back in to your session?

popey, 2020-05-02:
You're welcome. Thanks for the super fast response. Yes, restarting my session fixed it. Thanks.
[image]

closed 2020-05-02 by popey