corecoding/Vitals #140: "Some code tried to set a deprecated GObject property" warning on start

terencode, 2020-05-02:
**Describe the bug**
While the extension functionalities remain in a working state, a warning is printed each time the extension is loaded/enabled.
**To Reproduce**
1. Enable extension
2. Check the log with `journalctl /usr/bin/gnome-shell -r` and notice the following after the extension was enabled:
```
Some code tried to set a deprecated GObject property.
                                                   0 _patchContainerClass/containerClass.prototype.child_set() ["resource:///org/gnome/shell/ui/environment.js":34:12]
                                                   1 _patchContainerClass/containerClass.prototype.add() ["resource:///org/gnome/shell/ui/environment.js":43:17]
                                                   2 _createHotItem() ["/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js":228:25]
                                                   3 wrapper() ["resource:///org/gnome/gjs/modules/script/_legacy.js":82:26]
                                                   4 _drawMenu() ["/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js":313:17]
                                                   5 wrapper() ["resource:///org/gnome/gjs/modules/script/_legacy.js":82:26]
                                                   6 _init() ["/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js":75:13]
                                                   7 wrapper() ["resource:///org/gnome/gjs/modules/script/_legacy.js":82:26]
                                                   8 enable() ["/usr/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js":509:17]
                                                   9 _callExtensionEnable() ["resource:///org/gnome/shell/ui/extensionSystem.js":165:31]
                                                   10 loadExtension() ["resource:///org/gnome/shell/ui/extensionSystem.js":335:25]
                                                   11 _loadExtensions/<() ["resource:///org/gnome/shell/ui/extensionSystem.js":568:17]
                                                   12 collectFromDatadirs() ["resource:///org/gnome/shell/misc/fileUtils.js":27:16]
                                                   13 _loadExtensions() ["resource:///org/gnome/shell/ui/extensionSystem.js":547:18]
                                                   14 _enableAllExtensions() ["resource:///org/gnome/shell/ui/extensionSystem.js":577:17]
                                                   15 _sessionUpdated() ["resource:///org/gnome/shell/ui/extensionSystem.js":608:17]
                                                   16 init() ["resource:///org/gnome/shell/ui/extensionSystem.js":54:13]
                                                   17 _initializeUI() ["resource:///org/gnome/shell/ui/main.js":254:21]
                                                   18 start() ["resource:///org/gnome/shell/ui/main.js":143:4]
                                                   19 <TOP LEVEL> ["<main>":1:46]
```

**Desktop **
 - OS: ArchLinux 5.6.8
 - Gnome Version 3.36.2

corecoding, 2020-05-03:
I saw the error you spoke of. Do you have the develop branch loaded by chance? I just pushed a fix there.

terencode, 2020-05-03:
Yup, I can confirm this is fixed. Will leave open until it's merged to master.

corecoding, 2020-05-06:
merged to master

closed 2020-05-06 by corecoding