corecoding/Vitals #141: Vitals increases status bar padding of other elements

terencode, 2020-05-02:
**Describe the bug**

**To Reproduce**
Steps to reproduce the behavior:
1. Enable the extension
2. Notice the padding of the different element of gnome-shell status bar is increased

**Expected behavior**
The padding of the other status bar elements like the one from the appindicator extension on the left and the gnome-shell owns icons on the right should stay the same.

**Video**
[image]

**Desktop**
 - OS: ArchLinux 5.6.10
 - Gnome Version 3.36

corecoding, 2020-05-03:
This sounds like a duplicate of #136. Can you help me by disabling all your extensions, and with Vitals on, see which extension combined with Vitals makes this happen?

terencode, 2020-05-03:
It happens with all extensions disabled except Vitals.

corecoding, 2020-05-04:
That makes it more difficult to fix - like taking your car to the mechanic. They would need to hear the "noise". Can anyone else shed some light on how to duplicate this?

corecoding, 2020-05-05:
Just thought of something @terencode  - did you override any of your system fonts?

terencode, 2020-05-05:
I think so, here are my settings:
[image]

I tried to revert the scaling factor to 1 but I could still reproduce the issue, it's just less noticeable.

corecoding, 2020-05-06:
Can you do me a huge favor? Inside .local/share/gnome-shell/extensions/Vitals@CoreCoding.com/stylesheet.css there are a few CSS entries. Can you edit the file, remove one line at a time, save the file, and restart gnome? alt-f2 then type 'r' . Turning the extension on and off in the Extensions app may not reload the stylesheet. I am curious if one of these css statements is bleeding over.

terencode, 2020-05-06:
You are the one doing me a huge favor with trying to figure this out :)

The culprit is `#panel .panel-button .system-status-icon { margin: 0 4px 0 8px; padding: 0; }` !
Removing it and restarting gnome-shell makes everything looks normal again.

vesterlay, 2020-05-07:
I've the same issue on my end. No settings have been changed, manjaro gnome. 
[image]

corecoding, 2020-05-07:
@terencode @vesterlay can you two check out the develop branch and see if it is fixed?

vesterlay, 2020-05-07:
It's alright now.

corecoding, 2020-05-07:
I added some padding between the icon/text bundles in the panel. What do you think?

terencode, 2020-05-07:
Nice but now I have an extra icon that does not show any value (ellipsized):
[image]

Also would it be possible to force vitals to be the first when put to the right? 
Right now here is how it looks: 
[image]

I would like it to be on the very left, before the screenshot and clipboard manager extension.

corecoding, 2020-05-09:
That will be another feature request. :) Not sure if it can be done, but I want to focus on the bugs right now. Closing this ticket. Woohoo

closed 2020-05-09 by corecoding

terencode, 2020-05-09:
I had a quick look and it doesn't seem possible, Vitals already tries to be the first when set to the right panel... 
What would be needed is an extension or a native way to reorder those icons...