corecoding/Vitals #142: Battery No Data with /sys/class/power_supply/BAT1 available

inos-github, 2020-05-03:
Values for  Battery are not shown - No Data
however, the data is available in /sys/class/power_supply/BAT1
sys/class/power_supply/BAT0 is not existing

files in /sys/class/power_supply/BAT1:
alarm
capacity
capacity_level
charge_full
charge_full_design
charge_now
current_now
cycle_count
manufacturer
model_name
power
present
status
technology
type
uevent
voltage_min_design
voltage_now

System:
Huawei Matebook 14 D (AMD)
Ubuntu 20.04

corecoding, 2020-05-09:
In preferences, next to Monitor battery, I added a sub dialog to change the "Numerical Slot" of the battery. It defaults to 0. Please set it to 1 and let me know if your battery stats start working. I don't have a laptop right now, so I am developing in the dark. This will be in the develop branch, instructions are in the README.md in how to clone the repo.

corecoding, 2020-05-11:
Can anyone else verify that the battery stats are at least working (even if you have BAT0) - once we get this verified I can submit a new version to the gnome extension page

inos-github, 2020-05-11:
I've tested the dev branch but still 'No Data'.
However, substituting 'this._settings.get_string('battery-slot')' in line 247 in sensors.js by '1' does work. So probably the hashtag (#) shown in the settings dialog is part of the string and should be removed?

corecoding, 2020-05-11:
Thank you! This is interesting. The #1, #2, #3 etc are the values, but the get_string returns the key, which is 0, 1, 2... I’ll log the output and double check.

corecoding, 2020-05-11:
Can you try one more time? Make sure your local changes are removed and then do a *git pull* to get my changes. I was using get_string instead of get_int.

inos-github, 2020-05-11:
it's fixed, thank you!

closed 2020-05-11 by corecoding