corecoding/Vitals #143: Multiple Storage Locations

esa1975, 2020-05-05:
**Describe the new feature you'd like**
I looked through the issues and didn't see this suggestion. I have two disks in my laptop and would like to be able to see both. I can see that this might be a challenge to implement cleanly from a UI perspective. Ideally, I'd like to interact with each storage location in the same way it currently works.

**Additional context**
Please consider this as more of an idea than a request. I'm sure you have more interesting things to add and this is certainly not a high priority. It's not that big of a deal to change the path in preferences if I want to switch. Thanks!

antoniocarelli, 2020-08-11:
I'd like this too

starsforeveryone, 2023-01-05:
I would also like to request this feature. My PC has several drives and it would be awesome to be able to monitor their free storage space for video editing!

starsforeveryone, 2023-01-19:
Earth to developers? Developers, come in?

corecoding, 2023-01-20:
There is one developer, me. Vitals is alive and well, but this issue has just started to gain a little traction. This request is larger than it seems.

One way to implement this would be to have two or more Storage sections. This fundamentally changes the core code of Vitals because each section starts off from the _sensorIcons array.

Another way to implement this is to stuff all of the drive stats / info into the Storage option itself, but I think that may feel cluttered.

Anyhow, I am open to reviewing PRs from anyone wanting to add this support. Not saying I'm against it, just need to focus on my day job.

starsforeveryone, 2023-01-20:
OMG, I didn't know Vitals was a one-man project! oO

I also didn't want to push you, just wanted to ask if there was a possibility for this feature to be implemented. I completely understand now that this is "just" a hobby project. For that, it's awesome! It looks a lot more professional to be honest, which is why I at first mistook it for a bigger project. :)

Thanks for replying mate!

yannikthegenius, 2024-09-29:
+1 for this feature; would be very useful.
In KDE, this is possible via "System Monitor Sensors".

Thanks for making the extension!