corecoding/Vitals #146: can't it open in tweaks extensions

kkyflying, 2020-05-22:
**Describe the bug**
can't it open in tweaks extensions

**To Reproduce**
Steps to reproduce the behavior:
1. glone the code
2. restart gnome
3. open the tweaks

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04.4]
 - Gnome Version [3.82]

thank  you  ~

kkyflying, 2020-05-22:
[image]

corecoding, 2020-05-22:
What branch are you on? Can you give me the latest error you see in /var/log/syslog?

kkyflying, 2020-05-23:
l  am on the develop brach,
```
but nothing at cat /var/log/syslog

➜  Vitals@CoreCoding.com git:(develop) journalctl --since="`date '+%Y-%m-%d %H:%M'`" -f | grep Vitals
5月 23 10:09:59  gnome-software[3026]: failed to create an app for user/*/extensions-web/shell-extension/Vitals_CoreCoding.com/*
5月 23 10:11:04  gnome-shell[2606]: JS WARNING: [~/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js 159]: assignment to undeclared variable prefsButton
```

next l delete the developcode and  try to install by https://extensions.gnome.org/extension/1460/vitals/ ,
now is can open it 

thank you so much

corecoding, 2020-05-29:
Glad it is working now. The develop branches are sometimes volatile.

closed 2020-05-29 by corecoding