corecoding/Vitals #149: Make it possible to select which temperatures / fan speeds should be averaged.

bajo, 2020-06-03:
The average functionality for fan speeds and temperatures should have the option to select which of the available sensor values should be used for average calculations.

In my case I have 2 different situations in which the calculations are meaningless because of the reported values.

1. Fan speeds are reported as 0 when no fan is connected. Out of four possible fans I only utilize the CPU fan. No case fan, AIO fan or secondary CPU fan is installed. Thus all of their values are reported as 0 rpm. Averaging the fan speeds results in a meaningless number. I'm actually not sure when this average number would actually be meaningful even if all values above 0rpm, but I guess this was well discussed before.

2. The drivers I use to report fan speeds and temperatures (it87 or asus-wmi-sensors) report one of the temperatures as 216 degree Celsius. As I understand this is a problem of the driver implementation but they won't be fixed there.

**Additional context**

TimoFreiberg, 2020-09-24:
I have a similar issue, several sensors return 0°C and one even returns ~-200°C which skews the average a lot. In my case, ignoring values <=0 in the average calculation would fix the issue without requiring big interface changes.

A simple interface change could be to have a setting for "ignore values smaller than X and greater than Y" (where X and Y are both optional, or maybe set to 0°C and 200°C by default?)

corecoding, 2020-11-05:
Looks great! I am submitting an update now. It will take anywhere from a few minutes to a week to be approved. You never know with Gnome. :)

closed 2020-11-05 by corecoding