corecoding/Vitals #150: Battery drain

A-Hussein-M, 2020-06-08:
**Description**
Hi, I am experiencing a huge battery drain (around 30% battery life hit) even though it's set to update every 6 seconds.
I used other monitoring extensions such as `system-monitor` [https://github.com/paradoxxxzero/gnome-shell-system-monitor-applet](url) which doesn't have that much of an impact even when set to refresh every second.

**Screenshots**
Here are some CPU charts from Gnome monitor (Average CPU usage doesn't seem to increase that much but CPU cores are showing different behaviour) 
**Note:** system monitor refreshes every 1 second

**OFF**
[image]

**ON** with **1 second** refresh
[image]

**ON** with **5 seconds** refresh
[image]

**System info**
 - OS: Fedora 32
 - Gnome Version 3.36.3
 - CPU: Intel i5-8250U (4 cores, 8 threads)

_I am not sure what data should be included in the bug report, so please let me know if you need any further info, Thank you!_

A-Hussein-M, 2020-06-08:
with the other extension
[image]

**My configuration**
[image]

corecoding, 2020-06-09:
Thank you for this detailed post. What is weird to me, is that even when Vitals is set to 5 seconds, your CPU usage charts don't show it. I would expect to see peaks and valleys.

Can you try turning off (in preferences) each sensor group one by one? Network, Temp, Fan, Battery, Storage, until it is using less CPU?

A-Hussein-M, 2020-06-09:
After trying to enable only one sensor group at a time, I think the two problematic sensors are `Temperature`  and `Voltage`.
I have tested multiple times with multiple scenarios: enabling all but them, disabling all but them & enabling only one at at a time.

corecoding, 2020-06-11:
If you run the sensors command at the terminal, how quickly does it return? If you run this command:
while true; do sensors; sleep 1; done
do you see similar cpu usage?

A-Hussein-M, 2020-06-11:
`sensors` command executes in `1.22 secs`, here is the output
```
~> time sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +50.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +50.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +43.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +45.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +44.0°C  (high = +100.0°C, crit = +100.0°C)

amdgpu-pci-0100
Adapter: PCI adapter
vddgfx:      862.00 mV 
edge:         +39.0°C  (crit = +104000.0°C, hyst = -273.1°C)
power1:        2.54 kW (cap =  18.00 W)

acpitz-acpi-0
Adapter: ACPI interface
temp1:         +0.0°C  (crit = +119.0°C)
temp2:         +0.0°C  (crit = +119.0°C)
temp3:        +10.0°C  

pch_skylake-virtual-0
Adapter: Virtual device
temp1:        +41.0°C  

BAT0-acpi-0
Adapter: ACPI interface
in0:          12.37 V  

________________________________________________________
Executed in    1.22 secs   fish           external 
   usr time    0.22 millis  217.00 micros    0.00 millis 
   sys time   24.82 millis   54.00 micros   24.76 millis 

```

> If you run this command:
> while true; do sensors; sleep 1; done
> do you see similar cpu usage?

Yes.
Since CPU charts don't make sense to me and average CPU usage doesn't seem to increase that much, I measure the impact on battery life and the results are very constant: Running the command `while true; do sensors; sleep 1; done` causes about `60%` hit, which is even more than that caused by setting the extension's refresh interval to 1s (with only `temperature` and `voltage` enabled).
Weirder to me, even when browsing the web and constantly switching between applications (with average CPU usage around 10%, sometimes more) it still doesn't drain that fast!

A-Hussein-M, 2020-06-11:
I also tried uninstalling `lm_sensors`, rebooting the system then testing the extension again with only `temperature` and `voltage` enabled, it can still pull data and it is still draining the battery with at same rate.

corecoding, 2020-06-11:
the sensors command, and Vitals, both have the same technique. They read data from the /sys and /proc directories. sensors only shows Temp, Fan and Volt, which comes solely from /sys. The /sys directory looks like files, but they are actually hooked into the kernel, and every time you poll them, the kernel is involved. Because the while true... command caused CPU usage, this confirms it is not Vitals but one of your sensors / the kernel.

Also, because running the command with time shows it takes longer than 1 second, it means that if you set the timeout in vitals to 1 second, you will get a backup of that one sensor - piling up and causing more CPU usage.

I just tried running the while command above without sleep 1; which hammers the system.  My system got up to a max of 9% CPU, and it was running the sensors command about 10 times per second.

A-Hussein-M, 2020-06-11:
Thank you for the explanation.
Using other extensions to collect the same data didn't cause any drain which gave me the impression that it's Vitals issue, especially because I previously set the refresh interval to 5 or 6 seconds. I will disable the temperature sensor for now. Thanks again!

corecoding, 2020-06-16:
I am curious if anyone else out there in the community experiences this. If so, we may be able to turn this into a feature assuming the CPU usage increase is accompanied by a delay in the round trip when querying the sensor. For example, if you go to /sys/class/hwmon/hwmon# and cat the various sensors, I am guessing you will see one that returns much slower than the rest. This should also cause CPU increase. If you can confirm this, let me know and I will think about a possible fix.

A-Hussein-M, 2020-06-16:
> if you go to /sys/class/hwmon/hwmon# and cat the various sensors

After doing so, it turned out that the battery drain is caused by the external GPU (**amdgpu**) rather than CPU usage. The results were:
`cat /sys/class/hwmon/hwmon0/temp*_input #acpitz`    Executed in `3.17 millis`
`cat /sys/class/hwmon/hwmon3/temp*_input #pch_skylake`    Executed in `2.95 millis`
`cat /sys/class/hwmon/hwmon4/temp*_input #coretemp`    Executed in `3.21 millis`
**_while_**
`cat /sys/class/hwmon/hwmon2/temp*_input #amdgpu`    Executed in `632.25 millis`

Now it makes sense why CPU usage didn't show that much of a change even with the ridiculous battery hit _https://github.com/corecoding/Vitals/issues/150#issuecomment-642631556_
To confirm this, I tried two things:
_1- disabled the external GPU in BIOS, booted the system and enabled all sensors in Vitals:_ **No battery drain**
_2- enabled the GPU again, disabled Vitals and enabled the other extension (`system-monitor`) that I previously had no issues with (because it was set to only read `pch_skylake` sensor) but this time i set it to read `amdgpu-temp1`:_ **Huge battery drain**

A-Hussein-M, 2021-10-31:
I think this is fixed on newer kernels.

closed 2021-10-31 by A-Hussein-M