corecoding/Vitals #153: HDD temp

Arup65, 2020-06-28:
Please add hard drive temp via udisks. Very full fledged extension otherwise keep the good work up.
Thanks

corecoding, 2020-06-28:
Is the hdd temp you speak of available in the /sys/class/hwmon folders? 

Christopher Monahan

Arup65, 2020-06-28:
On extensions like Freon and others the access is via udisks or via hddtemp.

Strykar, 2020-07-11:
> Is the hdd temp you speak of available in the /sys/class/hwmon folders? Christopher Monahan Sent from my iPhone

IIRC, it's not exposed via sysfs because it needs to be read from SMART, and not all drives (in the past) supported SMART, hell, `hddtemp` can't pull temps off a SATA SSD, forget NVME.

It's why Freon groks it from `hddtemp`/`smartctl`/`udisks2`.

corecoding, 2022-07-10:
Per the WiKi, Vitals reads from /proc and /sys to prevent system lag. I would entertain a PR if someone wanted to do the dev work, but a disclaimer would have to be added in Preferences explaining the system lag and not to report a bug for said lag.

closed 2022-07-10 by corecoding